JMRI/JMRI #9890: LogixNG TODO

dsand47, 2021-06-13:
- [ ] Implement **intensity** and **transition time** for the **Light** action.
- [ ] Update the **Light** Logix import.
- [ ] Implement the **Audio** action.
- [ ] Create the **Audio** Logix import.
- [ ] Migrate the technical help pages.

dsand47, 2021-07-09:
LogixNG is production with the 4.24 release.

closed 2021-07-09 by dsand47

danielb987, 2021-07-09:
@dsand47 
I'm fine with this issue being closed. But my question is what's the state of the technical pages of LogixNG? For example how to add a new action or expression?

I think it would be good to have a link `LogixNG` under the section `Code Structure` on this page:
https://www.jmri.org/help/en/html/doc/Technical/index.shtml

One of the most important goals for me when I created LogixNG was to make it easy for others to extend it, to add new stuff to it. Actions, expressions and formula functions.

danielb987, 2021-07-09:
There is also a function missing:

**readReference(ref)**

Returns an Object
`ref` is the reference, for example `{IM1}` or `Yard table[Left turnout,North yard]`

With this function, you can use a LogixNG Table for ordinary data, for example a time table. I thought about this when I read [this thread](https://groups.io/g/jmriusers/message/191964) on the JMRI users list.

dsand47, 2021-07-09:
> @dsand47
> I'm fine with this issue being closed. But my question is what's the state of the technical pages of LogixNG? For example how to add a new action or expression?
>
> I think it would be good to [...]

The third section of the Reference chapter list is going to be the technical chapters.

dsand47, 2021-07-09:
> There is also a function missing:
>
> **readReference(ref)**
>
> Returns an Object
> `ref` is the reference, for example `{IM1}` or `Yard table[Left turnout,North yard]`
>
> With this function, you can use a [...]

What do you mean by "missing"?  The function help has a `readMemory` but I don't find a `readReference`.

danielb987, 2021-07-09:
> What do you mean by "missing"? The function help has a `readMemory` but I don't find a `readReference`.

Exactly that. I have not yet implemented it. Like a lot of other missing functions, like cos() and atan().

danielb987, 2021-07-09:
> The third section of the Reference chapter list is going to be the technical chapters.

Thanks!

danielb987, 2021-07-09:
> Exactly that. I have not yet implemented it. Like a lot of other missing functions, like cos() and atan().

Which brings the question which functions should be implemented? For example the Java Math class has a lot of functions that probably will never be used in LogixNG.