JMRI/JMRI #9891: OpenLCB turnout create fails with leading zero on first event

KenC57, 2021-06-13:
The input validation (while you type) accepts the values as copied from most uses of the event ids and shows green that it is ok. The hover text correctly shows the form of the events using leading/padded zeros.

Something has changed deeper (after hitting the create) that causes that form to be rejected. My example input was:
01.02.03.04.05.06.07.08;01.02.03.04.05.06.07.09

Not sure which version introduced this bug. But 4.23.6 has it.

KenC57, 2021-06-13:
The form in the OpenLCB Turnout tests never try with leading zeros. In fact all tests us single digits in each byte of the address.

RobertPHeller, 2021-06-27:
Since at least 4.20.  I'm downloading the source and have a look...

RobertPHeller, 2021-06-27:
> The form in the OpenLCB Turnout tests never try with leading zeros. In fact all tests us single digits in each byte of the address.

Well, that was dumb.. The most common real-world usage is certain to have leading zeros!

RobertPHeller, 2021-06-27:
After a very little bit groveling in the source code and inserting some printf()s it appears that the system names are getting double prefixed:

      [java] *** Manager.makeSystemName(MS02.01.57.00.01.A2.00.66;02.01$
     [java] *** OlcbAddress.svalidateSystemNameFormat(M2SMS02.01.57.00$
     [java] *** OlcbAddress.svalidateSystemNameFormat() name is 'M2SMS$
     [java] *** OlcbAddress.validateSystemNameFormat(M2SMS02.01.57.00.$
     [java] *** OlcbAddress.validateSystemNameFormat(): prefix is 'M2S$
     [java] *** OlcbAddress.validateSystemNameFormat(): oAddr = 'MS02.$
     [java] *** OlcbAddress.java(MS02.01.57.00.01.A2.00.66;02.01.57.00$
     [java] *** OlcbAddress.split(): aString = MS02.01.57.00.01.A2.00.$
     [java] *** OlcbAddress.split(): pStrings.length is 2

RobertPHeller, 2021-06-27:
This is using the "Make Sensor" functionality in the CDI configuration tool, not from the PanelPro Sensor table Add button, but I expect there is a similar problem -- there seems to be problem about who adds the prefix and what the prefix is to be -- you will note that not only are there two prefixes, but that they are *different* prefixes...

dsand47, 2021-06-27:
The M2S prefix indicates the second CAN connection.  The first connection would be MS.  

It is possible that the CDI tool is assuming one CAN connection which would be MS.

RobertPHeller, 2021-06-27:
> The M2S prefix indicates the second CAN connection. The first connection would be MS.
> 
> It is possible that the CDI tool is assuming one CAN connection which would be MS.

No, M2S indicates a two event sensor.  The MS prefix is exranous.

RobertPHeller, 2021-06-27:
Here is the fix.  It appears that the OpenLCB ClientActions is adding in its own prefix, that the rest of program does not expect or know to deal with.

[ClientActions.patch.txt](https://github.com/JMRI/JMRI/files/6722283/ClientActions.patch.txt)

bobjacobsen, 2021-06-27:
> No, M2S indicates a two event sensor. The MS prefix is exranous.

Then that’s your problem.

JMRI prefixes require a parseable format: A letter, zero or more digits, then a type letter.  In this case, the letter & digits connection prefix is M2, and the type letter is S.

Usually M2 indicates a 2nd connection, with M as a first one (i.e. MS for sensors on the first and M2S for sensor on the 2nd).  But it doesn’t have to.

If the OpenLCB code is doing something else, it’s wrong.

RobertPHeller, 2021-06-27:
> 
> 
> > No, M2S indicates a two event sensor. The MS prefix is exranous.
> 
> Then that's your problem.
> 
> JMRI prefixes require a parseable format: A letter, zero or more digits, then a type letter.  In this case, the letter & digits connection prefix is M2, and the type letter is S.
> 
> Usually M2 indicates a 2nd connection, with M as a first one (i.e. MS for sensors on the first and M2S for sensor on the 2nd).  But it doesnât have to.
> 
> If the OpenLCB code is doing something else, it's wrong.

Yes. The ClientAction code (the GUI code) is inserting one prefix, and later
the sensor manager code is inserting another prefix and passing that along to
the address parsing code, which (internally) strips the prefix added by the
manager code and tries to parse the event IDs, but dies because of the prefix
inserted by the GUI code. That is, the sensor (or turnout) system name has
*two* *different* prefixes, one put in by the GUI code and another one put in
by the manager code. The manager code does check to see if the prefix in
there, but it is checking for a different prefix than what the GUI code was
putting on, so it perpends its own. *Maybe* the GUI code is just putting on
the wrong prefixes. I just commented the piece of code out (rather then just
delete it), in case it makes sense to fix it to put in the correct prefixes.
Either way should fix things.

> 
> 
> 
> https://github.com/JMRI/JMRI/issues/9891#issuecomment-869222347

bobjacobsen, 2021-06-27:
`SensorManager` and `TurnoutManager` instances know about the proper prefix for a connection; the GUI (usually) doesn't unless it's gotten it from a `Manager` or some other connection instance.

I asked about the content of the 2nd variable on the PR because of this.  It seems likely that if the 1st one has a prefix, the 2nd one does too. And that'll eventually be an error.

RobertPHeller, 2021-06-27:
At Sun, 27 Jun 2021 14:27:03 -0700 JMRI/JMRI ***@***.***> wrote:

> 
> `SensorManager` and `TurnoutManager` instances know about the proper prefix
> for a connection; the GUI (usually) doesn't unless it's gotten it from a
> `Manager` or some other connection instance.

Right. The GUI should not be inserting prefixes on its own.  That is the job 
of the SensorManager and TurnoutManager. 

> I asked about the content of the 2nd variable on the PR because of this. It
> seems likely that if the 1st one has a prefix, the 2nd one does too. And
> that'll eventually be an error.

Neither (should) have a prefix. The "address" is two event ids separated by a
semicolon. The OlbcAddress class understands how to deal with that.

> 

-- 
Robert Heller             -- Cell: 413-658-7953 GV: 978-633-5364
Deepwoods Software        -- Custom Software Services
http://www.deepsoft.com/  -- Linux Administration Services
***@***.***       -- Webhosting Services

KenC57, 2021-06-27:
I had logged the earlier bug as finding it via the JMRI table Make Sensor. There it was showing the leading
zero as the correct format (green flag in entry box) but after hitting Create, it came back with the error.

bobjacobsen, 2021-06-27:
> Neither (should) have a prefix. The "address" is two event ids separated by a
> semicolon. The OlbcAddress class understands how to deal with that.

Then why does adding one prefix result in two prefixes?

dsand47, 2021-06-27:
> Neither (should) have a prefix. The "address" is two event ids separated by a semicolon. The OlbcAddress class understands how to deal with that.
> Then why does adding one prefix result in two prefixes?

My understanding is that if the connection prefix is M2, then MSevent:event becomes M2SMSevent:event via provideSensor().  This assumes the the M2 connection is first connection and therefore the default when a valid connection prefix is not supplied.

RobertPHeller, 2021-06-28:
At Sun, 27 Jun 2021 15:25:55 -0700 JMRI/JMRI ***@***.***> wrote:

> 
> > Neither (should) have a prefix. The "address" is two event ids separated by a
> > semicolon. The OlbcAddress class understands how to deal with that.
> 
> Then why does adding one prefix result in two prefixes?

Two *different* prefixes. The GUI code was adding "MS" (or "MT"). The
SensorManager and TurnoutManager instances were checking for and adding a
*different* prefix, "M2S" (or "M2T"). By the time it gets to OlbcAddress, the
string is

M2SMS02.03.04.05.06.07.08;20.30.40.50.60.70.80

OlbcAddress strips off the 'M2S' then splits at the ';', yielding:

'MS02.03.04.05.06.07.08' and '20.30.40.50.60.70.80' then splits on the dots, 
and then does a ParseInt on 'MS02' and fails.

Either the GUI code should prepend NO prefix (my fix) or propend the *correct* 
prefix. The SensorManager and TurnoutManager instances check for 'M2S' (or 
'M2T') before adding another prefix. 

> 
> 
> 
> 

-- 
Robert Heller             -- Cell: 413-658-7953 GV: 978-633-5364
Deepwoods Software        -- Custom Software Services
http://www.deepsoft.com/  -- Linux Administration Services
***@***.***       -- Webhosting Services

RobertPHeller, 2021-07-05:
> The input validation (while you type) accepts the values as copied from most uses of the event ids and shows green that it is ok. The hover text correctly shows the form of the events using [...]

OK, I found something interesting with OlcbSensorManager.getNextValidAddress() and OlcbTurnoutManager.getNextValidAddress().

OlcbSensorManager.getNextValidAddress() does nothing but return its first argument and OlcbTurnoutManager.getNextValidAddress() passes its first argument *without* the prefix to OlcbAddress.validateSystemNameFormat(), which *fails*.  I don't know which is wrong.  Possibly *both* are wrong, (the Olcb sensor manager  is ommiting a test and Olcb turnout manager is doing the test wrong...

dsand47, 2021-07-05:
@RobertPHeller 

If you go to the sensor table in JMRI, select the connection from the connection tab and do an add with

01.02.03.04.05.06.07.08;01.02.03.04.05.06.07.09

does it create the sensor table entry with the "connection prefix" + "S" added to the beginning?  Look at Preferences &rArr; Connections to get the connection prefix.

RobertPHeller, 2021-07-05:
> @RobertPHeller
> 
> If you go to the sensor table in JMRI, select the connection from the connection tab and do an add with
> 
> 01.02.03.04.05.06.07.08;01.02.03.04.05.06.07.09
> 
> does it create the sensor table entry with the "connection prefix" + "S" added to the beginning? Look at Preferences ⇒ Connections to get the connection prefix.

That is not the problem.  The sensor is created properly. The problem is *specificly* with the *getNextValidAddress()* methods.  The OlcbSensorManager version peforms no checks (should it?) and the turnout version does it wrong (or maybe it should not be checking? -- there is a comment suggesting that it should not be checking and then it does and does  it wrong).  Once things are validated (elsewhere) the system name is created properly.  I don't know enough about JMRI workflow to know what the proper fix should be.

dsand47, 2021-07-05:
Are you saying the double prefix and leading zero issues have been resolved?

Here is another test with the connection prefix set to M:
```
>>> import java
... import jmri
... 
... s = sensors.provideSensor('01.02.03.04.05.06.07.01;01.02.03.04.05.06.07.02')
... 
... print 'sensor = {}'.format(s)
sensor = MS01.02.03.04.05.06.07.01;01.02.03.04.05.06.07.02
```

RobertPHeller, 2021-07-05:
> Are you saying the double prefix and leading zero issues have been resolved?

They are *separate* issues.  The double prefix was caused by having hard-wired prefix code in the GUI code (which went unnoticed until I opened a *second* LCC connection which resulted in a *different* prefix).  Solving this is to grab the real (dynamic) prefix in the GUI code.

The leading zero issue (which might be misnamed) is caused by some weirdness with OlcbTurnoutManager's coding of its getNextValidAddress() method, which is coded differently from OlcbSensorManager's getNextValidAddress() method -- I suspect that possibly both are wrong, but in different ways.  Either OlcbTurnoutManager's getNextValidAddress() method should be prepending the prefix or not be calling OlcbAddress.validateSystemNameFormat() at all and maybe OlcbSensorManager's getNextValidAddress() should be doing something more than it is (maybe also calling  OlcbAddress.validateSystemNameFormat() with the prefix prepended).  In any case,  Manager.makeSystemName(), does call OlcbAddress.validateSystemNameFormat() eventually (actually calls it multiple times).  I've attached snipets of the log files [I put in debug code with stack traces].
[log-Turnout.txt](https://github.com/JMRI/JMRI/files/6765095/log-Turnout.txt)
[log-Sensor.txt](https://github.com/JMRI/JMRI/files/6765099/log-Sensor.txt)

> 
> Here is another test with the connection prefix set to M:
> 
> ```
> >>> import java
> ... import jmri
> ... 
> ... s = sensors.provideSensor('01.02.03.04.05.06.07.01;01.02.03.04.05.06.07.02')
> ... 
> ... print 'sensor = {}'.format(s)
> sensor = MS01.02.03.04.05.06.07.01;01.02.03.04.05.06.07.02
> ```

KenC57, 2021-07-05:
My suspicion of the leading zero is general code doing trimming on user entry. Some systems required the
trimming of leading zeros while other systems, like LCC, do not and in normal notation in LCC preserve the
leading zeros. I recall during the last year some other system connections needed fixing and I think it
impacted the LCC side of things. I haven't been able to trace any sort of change like this inside the OpenLCB
side of the code which is why I suspect something in the general JMRI handling is the issue.

RobertPHeller, 2021-07-05:
At Mon, 05 Jul 2021 08:55:21 -0700 JMRI/JMRI ***@***.***> wrote:

> My suspicion of the leading zero is general code doing trimming on user entry. Some systems required the
> trimming of leading zeros while other systems, like LCC, do not and in normal notation in LCC [...]

Actually, no this is not the problem.  It is the getNextAddress() methods of 
the Olbc managers.  The Sensor manager does nothing (and things work properly) 
and the Turnout version tries to validate the address, without a prefix, which 
fails.

It in unclear if the Sensor/Turnout managers should or should not be validing 
the address in their getNextAddress() methods or not.  I expect that if they 
are, they need to put the prefix on.  Their createSystemName() methods might 
also be wrong, but these methods are not actually called(!).

I'm thinking that the authors of JMRI proper and the authors of Openlcb.java 
might not be on the same page...

> 
> 
> 

-- 
Robert Heller             -- Cell: 413-658-7953 GV: 978-633-5364
Deepwoods Software        -- Custom Software Services
http://www.deepsoft.com/  -- Linux Administration Services
***@***.***       -- Webhosting Services

dsand47, 2021-07-05:
The Olcb maangers override the AbstractManager implementation of getNextAddress since that logic cannot work with the LCC address format.  The validation in the turnout manager seems redundant since curAddress is being returned unchanged.

KenC57, 2021-07-08:
Using 4.23.8 to confirm it is still a problem. I think this is significant as normal cut/paste of events will fail in the turnout table. Almost all events right now have a leading zero.
I just tried to create a turnout and a sensor using the string:
01.02.03.04.05.06.07.08;01.02.03.04.05.06.07.09
It created the sensor just fine. The turnout threw a pop up:
[image]
Note that the little green dot in the data entry field shows the UI thinks the input is valid.
The connection was OpenLCB simulator.

RobertPHeller, 2021-07-08:
Yes, what is happening is OlcbTurnoutManager's getNextAddress() method is passing the raw address to OlcbAddress.validateSystemNameFormat(), which really wants a *system name*, not a raw address, so it fails.  OlcbSensorManager's getNextAddress() method does nothing except return it argument, so creating sensors work.

bobjacobsen, 2022-05-13:
See discussion in #9921

bobjacobsen, 2023-03-29:
I've read through this, but I don't understand how to recreate the problem. I haven't been able to with the current head of master, so I'm probably not trying the right thing.

Could somebody please give a step-by-step way to show the problem, preferably on recent code?

Thank you.

KenC57, 2023-03-29:
My review of the history would say the title is a bit off. The issue was that
everything showed the events padded with zeros and that the create was having
issues with it. I think many different things have been done since then and this
may have been resolved. But a test case of using and not using the leading zero
for both the first and second event in the input would be the test. So:
Neither with leading zero
First with leading zero
Second with leading zero
Bother with leading zero

Do all of these work now?

bobjacobsen, 2023-03-29:
PR #11888 adds a test to make sure this is, and stays, resolved.

closed 2023-03-29 by bobjacobsen

RobertPHeller, 2023-03-29:
My understanding is that there is a problem between what the GUI is doing and 
the underlying data structs.  JMRI is trying to handle all possible connection 
types and somewhere got lost.  Actually the problem showed up for me when I 
had multiple LCC connections -- each connection gets its own prefix (why?) and 
this is what caused the problem.  Part of the problem was a matter of which 
piece of code included the prefix and which did not and how long was the 
prefix (the second/third connection had a longer prefix -- opps!).

At Wed, 29 Mar 2023 03:33:16 -0700 JMRI/JMRI ***@***.***> wrote:

> My review of the history would say the title is a bit off. The issue was that
> everything showed the events padded with zeros and that the create was having
> issues with it. I think many different [...]

-- 
Robert Heller             -- Cell: 413-658-7953 GV: 978-633-5364
Deepwoods Software        -- Custom Software Services
http://www.deepsoft.com/  -- Linux Administration Services
***@***.***       -- Webhosting Services

RobertPHeller, 2023-03-29:
It is possible that things have been fixed somewhere along the line.  

At Wed, 29 Mar 2023 06:26:35 -0700 JMRI/JMRI ***@***.***> wrote:

> 
> TLDR: I can't make this fail now, even with three separate LCC connections.  Would appreciate some pointers to how to do that with current code.
> 
> For some background on names, see the [naming page](https://www.jmri.org/help/en/html/doc/Technical/Names.shtml).
> 
> I created three OpenLCB connections with system prefixes M, M2 and O.  I then went to the Turnout Table and used the Add... window to try the following.  All worked as expected:
> 
>  - With the M prefix connection selected, I tried to add turnouts with eventIDs that started with zero, had leading zero on a non-zero 1st byte, and had just a non-zero first byte.
>  - Similarly with the M2 connection selected
>  - Similarly with the O connection selected.
> 
> These all worked as expected. They created turnouts with the expected system names, and the turnouts worked with the proper events on the proper connection.
> 
> There was one thing that might be worth changing.  As I typed in the names, there were sequential ERROR messages logged. For example:
> ```
>      [java] 09:07:53,992 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 1, address is 1. [AWT-EventQueue-0]
>      [java] 09:07:53,993 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 1, address is 1. [AWT-EventQueue-0]
>      [java] 09:07:54,271 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 2, address is 1.2 [AWT-EventQueue-0]
>      [java] 09:07:54,272 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 2, address is 1.2 [AWT-EventQueue-0]
>      [java] 09:07:54,434 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 2, address is 1.2. [AWT-EventQueue-0]
>      [java] 09:07:54,434 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 2, address is 1.2. [AWT-EventQueue-0]
>      [java] 09:07:54,542 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 3, address is 1.2.3 [AWT-EventQueue-0]
>      [java] 09:07:54,542 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 3, address is 1.2.3 [AWT-EventQueue-0]
>      [java] 09:07:54,741 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 3, address is 1.2.3. [AWT-EventQueue-0]
>      [java] 09:07:54,741 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 3, address is 1.2.3. [AWT-EventQueue-0]
>      [java] 09:07:54,876 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 4, address is 1.2.3.4 [AWT-EventQueue-0]
>      [java] 09:07:54,876 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 4, address is 1.2.3.4 [AWT-EventQueue-0]
>      [java] 09:07:55,035 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 4, address is 1.2.3.4. [AWT-EventQueue-0]
>      [java] 09:07:55,035 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 4, address is 1.2.3.4. [AWT-EventQueue-0]
>      [java] 09:07:55,124 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 5, address is 1.2.3.4.5 [AWT-EventQueue-0]
>      [java] 09:07:55,125 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 5, address is 1.2.3.4.5 [AWT-EventQueue-0]
>      [java] 09:07:55,307 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 5, address is 1.2.3.4.5. [AWT-EventQueue-0]
>      [java] 09:07:55,307 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 5, address is 1.2.3.4.5. [AWT-EventQueue-0]
>      [java] 09:07:55,427 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 6, address is 1.2.3.4.5.6 [AWT-EventQueue-0]
>      [java] 09:07:55,427 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 6, address is 1.2.3.4.5.6 [AWT-EventQueue-0]
>      [java] 09:07:55,563 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 6, address is 1.2.3.4.5.6. [AWT-EventQueue-0]
>      [java] 09:07:55,563 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 6, address is 1.2.3.4.5.6. [AWT-EventQueue-0]
>      [java] 09:07:55,724 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 7, address is 1.2.3.4.5.6.7 [AWT-EventQueue-0]
>      [java] 09:07:55,724 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 7, address is 1.2.3.4.5.6.7 [AWT-EventQueue-0]
>      [java] 09:07:55,834 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 7, address is 1.2.3.4.5.6.7. [AWT-EventQueue-0]
>      [java] 09:07:55,835 jmri.jmrix.openlcb.OlcbAddress        ERROR - unexpected number of terms: 7, address is 1.2.3.4.5.6.7. [AWT-EventQueue-0]
> ```
> These should probably be changed to DEBUG or even TRACE, as they're quite normal when a partial value has been entered.
> 
> 
> 
> 

-- 
Robert Heller             -- Cell: 413-658-7953 GV: 978-633-5364
Deepwoods Software        -- Custom Software Services
http://www.deepsoft.com/  -- Linux Administration Services
***@***.***       -- Webhosting Services

RobertPHeller, 2023-03-29:
At Wed, 29 Mar 2023 09:59:14 -0400 (EDT) Robert Heller ***@***.***> wrote:

> 
> It is possible that things have been fixed somewhere along the line.  

Yes: Fix turnout table add LCC turnout bug #9943 

> At Wed, 29 Mar 2023 06:26:35 -0700 JMRI/JMRI ***@***.***> wrote:

-- 
Robert Heller             -- Cell: 413-658-7953 GV: 978-633-5364
Deepwoods Software        -- Custom Software Services
http://www.deepsoft.com/  -- Linux Administration Services
***@***.***       -- Webhosting Services