JMRI/JMRI #9895: Travis jobs not reporting status

dsand47, 2021-06-15:
Could this be the problem:

[image]

bobjacobsen, 2021-06-18:
There's a required migration.  I've tried to do it several times now without success.  It has worked for other repositories I work on, but not the JMRI/JMRI and JMRI/website ones.  It seems to have something to do with our use of GitHub Actions, but I haven't found anybody who knows what's going on.

See
https://docs.travis-ci.com/user/migrate/open-source-repository-migration

Also
https://devops.stackexchange.com/questions/1201/whats-the-difference-between-travis-ci-org-and-travis-ci-com/4305
https://www.jeffgeerling.com/blog/2020/travis-cis-new-pricing-plan-threw-wrench-my-open-source-works

As an alternative:
https://docs.github.com/en/actions/learn-github-actions/migrating-from-travis-ci-to-github-actions

danielb987, 2021-06-18:
I think migrating from Travis to GitHub actions would be a good thing to do.

danielb987, 2021-08-02:
Travis is not used anymore by JMRI. But it would be good to keep this issue open since it would be beneficial if we could improve our test suites. I'm working on it, but I haven't been successful.

closed 2021-11-16 by stale[bot]