JMRI/JMRI #9919: Error during install of ver 4.22

HarlanSub, 2021-06-24:
While installing ver 4.22 over my existing 4.20 I got the following error "Error opening file for writing: C:\Program Files (x86)\JMRI\LaunchJMRI.exe         Click abort to stop the installation, Retry to try again, or Ignore to skip this file.   I tried retry several times with no luck, so I hit Ignore.  The program completed installation and it appears to be performing normally.   I just wonder if there is some lurking issue beneath the surface that I haven't encountered yet.  JMRI was running fine before the upgrade.   The computer is running Windows 10 Pro.

[image]

danielb987, 2021-06-26:
Welcome to JMRI and thanks for your report!

Have you asked this question on the JMRI users list? There are much more experienced people following that list than there are reading the Issues.
[Join JMRI users list](https://groups.io/g/jmriusers/join)
[BrowseJMRI users list](https://groups.io/g/jmriusers/topics)

If you join the JMRI users list and post a message, it might take some days before the moderators has time to approve your message.

MarcFour, 2021-06-28:
You possibly had JMRI running during your attempt to install newer version.  That or JMRI did not shutdown proper prior to the update.  LaunchJMRI is used to startup JMRI be it DecoderPro or PanelPro.  The install process wants to write to a file which is in an " open state" hence the warning.

HarlanSub, 2021-06-29:
Thanks for your response. I suspect that you are correct as occasionally when I start JMRI it tells me that it is already running. However in task manager(Windows 10), it shows that no applications are running.   

Should I reinstall the program or take other corrective actions?

Tim Mann

MarcFour, 2021-06-29:
Since, as you state, the program seems to be operational, I would not bother and just continu on.

I forced  the same error on my Win10 Pro by leaving JMRI running and going thru the reinstall of JMRI 4.22. It posted exactly the same error message.  I simply "ignored" (as you did)   with no ill effect.

HarlanSub, 2021-06-29:
Thank you. 

Tim Mann

closed 2021-08-13 by danielb987