JMRI/JMRI #9921: Can't create a sensor or turnout with the first event number starting with 0

RobertPHeller, 2021-06-27:
When I try to make a JMRI Sensor from *legitimate* events from a LCC node (a Tower-LCC), JMRI fails to create the sensor, claiming "ERROR - Invalid system name for Sensor: Did not find usable system name: M2SMS02.01.57.00.01.A2.00.66;02.01.57.00.01.A2.00.67 to a valid Olcb sensor address [AWT-EventQueue-0]

This is with JMRI.4.22+R4c4384d74 (latest production version).
[image]
[Error.log](https://github.com/JMRI/JMRI/files/6721752/Error.log)

KenC57, 2021-06-27:
THis is a duplicate of #9891

RobertPHeller, 2021-07-05:
> THis is a duplicate of #9891

Actually not a duplicate.  It is something else.  The GUI code creating PanelPro table entries (create Turnout/Sensor) is just broken WRT connection prefix handling).  9891 (leading zero weirdness) is something completely different.

bobjacobsen, 2021-07-06:
> When I try to make a JMRI Sensor from legitimate events from a LCC node (a Tower-LCC), JMRI fails to create the sensor, claiming "ERROR - Invalid system name for Sensor: Did not find usable system [...]

Could you please provide more detail on exactly what you do to “try to make a JMRI sensor”.  Enough detail that somebody can reproduce this would really help.  That includes how your preferences are configured: How many system connections do you have configured, and what are their types and letters?

RobertPHeller, 2021-07-06:
> Could you please provide more detail on exactly what you do to âtry to make a JMRI sensorâ.  Enough detail that somebody can reproduce this would really help.  That includes how your preferences are configured: How many system connections do you have configured, and what are their types and letters?

It happens when I have a 2nd LCC connection (M2 by default).  Appearently, the 
ClientActions.java code was hard wired to prefix its addresses with 'M' and 
the type letter ('S' or 'T'), but if things were set up with a 2nd LCC 
connection (M2 by default), code later was matching with M2 and the type 
letter, and because the match failed, was prepending a second prefix.  When 
the addresses was validated, it took off the second prefix and tried to 
validate the address with the first prefix still on, causing the validation to 
fail.  The fix is for the ClientActions.java get the *proper* prefix from 
proper place.

KenC57, 2021-07-06:
Look at #9891 it has my simple test case before Robert brought up what he found (connection prefix issue).

KenC57, 2021-07-08:
Using 4.23.8.
I just tried to create a turnout and a sensor using the string:
01.02.03.04.05.06.07.08;01.02.03.04.05.06.07.09
It created the sensor just fine. The turnout threw a pop up:
[image]
The connection was OpenLCB simulator.

RobertPHeller, 2021-07-08:
At Thu, 08 Jul 2021 05:44:04 -0700 JMRI/JMRI ***@***.***> wrote:

> 
> Using 4.23.8.
> I just tried to create a turnout and a sensor using the string:
> 01.02.03.04.05.06.07.08;01.02.03.04.05.06.07.09
> It created the sensor just fine. The turnout threw a pop up:
> [image]
> The connection was OpenLCB simulator.

Creating [LCC] turnouts from the Tables->Turnouts is a problem, due to a
problem with OlcbTurnoutManager's getNextValidAddress() is miscoded and calls
OlcbAddress.validateSystemNameFormat() with an invalid system name -- it
passes the raw address to OlcbAddress.validateSystemNameFormat(). This is
fixed in PR #9943.

The getNextValidAddress in OlcbSensorManager is a noop, so there is no error.

(The getNextValidAddress method makes no sense for LCC, since LCC does not 
support sequencial addresses.)

-- 
Robert Heller             -- Cell: 413-658-7953 GV: 978-633-5364
Deepwoods Software        -- Custom Software Services
http://www.deepsoft.com/  -- Linux Administration Services
***@***.***       -- Webhosting Services

dsand47, 2021-07-08:
> At Thu, 08 Jul 2021 05:44:04 -0700 JMRI/JMRI ***@***.***> wrote:
> Using 4.23.8. I just tried to create a turnout and a sensor using the string: 01.02.03.04.05.06.07.08;01.02.03.04.05.06.07.09 It [...]

Since getNextValidAddress is a no-op for both sensor and turnout, why do a createSystemName that is never used?

RobertPHeller, 2021-07-08:
I was trying to be consistent with other sensor and turnout managers for other connection types.  Just in case someday getNextValidAddress stops being a no-op.  I don't think createSystemName is actually called anyplace else -- the original version in both managers was wrong and would have crashed if called.

KenC57, 2021-08-23:
It is important that the form of the event for uses match. Rejecting the leading zero causes confusion.

RobertPHeller, 2021-08-23:
The problems with both problems (there are two *different* bugs) is not
anything to do with leading zeros, but with system prefix handling in either
the OpenLCB UI code or the OpenLCB -manager code. The former had the system
prefix "hardwired", which works so long as there is only one LCC connection
that uses the default prefix, if there are multiple LCC connections and/or the
system/connection prefix is not the default, the UI uses the wrong prefix and
things break. With the latter, the OpenLCB turnout manager code was not
properly handling the system prefix -- it was trying to verify an address pair
without a prefix with a method that expected a system name (eg with a prefix)
-- the left hand had no clue what the right hand was doing...  (The address 
verification is actually unneeded, but it is part of the API for support for 
sequencial addressing, which LCC does not support and is disabled in the GUI.)

There are fixes for both problems, and I guess they are in 4.24. 

At Mon, 23 Aug 2021 04:31:05 -0700 JMRI/JMRI ***@***.***> wrote:

> 
> It is important that the form of the event for uses match. Rejecting the leading zero causes confusion.
> 

-- 
Robert Heller             -- Cell: 413-658-7953 GV: 978-633-5364
Deepwoods Software        -- Custom Software Services
http://www.deepsoft.com/  -- Linux Administration Services
***@***.***       -- Webhosting Services

KenC57, 2021-08-23:
Verified as fixed in 4.24 with respect to leading zeros. Other invalid inputs still cause a stack trace without explaining that to the user. But deserves a new issue.

closed 2021-08-23 by KenC57