JMRI/JMRI #9932: Last few lines not displaying on web pages

jerryg2003, 2021-06-30:
Just noticed (but may have been there before), that I can't get the last few lines of web pages on my Dell laptop monitor under Windows 10/Chrome browser.  [I thought this was reported somewhere before but I couldn't locate it - apologies if so]  Scroll bar just kind of gets lost at bottom of window (happens both when window is partial screen and when full screen).

On ipad/Safari, similar behavior but I can pull the screen up by touch to see the last few lines.  Once I let go, those lines are lost again.

[image]

dsand47, 2021-06-30:
Looks like it needs the same fix to Footer.shtml as was done for SidebarTail.shtml, using 3 nearly empty paragraphs.

dsand47, 2021-06-30:
> Looks like it needs the same fix to Footer.shtml as was done for SidebarTail.shtml, using 3 nearly empty paragraphs.

I suspect that actual problem is that the **_side_** and **_mainContent_** divs are defined as **_fixed_** with their top at 64px.  This keeps the content below the header bar.  To enable scrolling, the **_max-height_** is set to 100% with **_overflow_** set to auto.  Maybe this causes the scroll are to extend 64px beyond the bottom of the window.

A recent change corrected a problem where mainContent was scrolling under the header bar.  This resulted in "on page" links jumping based on the top being at 0 which made the visible content start 64px beyond the target.  This is probably when the bottom of the content became unreachable.

Adding 3 paragraphs with `&nbsp;` creates a 84px blank area at the bottom.

silverailscolo, 2021-07-02:
_Adding 3 paragraphs with &nbsp; creates a 84px blank area at the bottom._
I think it would be better to use css for this: no need to edit (all) pages and in principle layout shouldn't be done through html.

dsand47, 2021-07-02:
The lines are added to Footer.shtml.

The **side** and **mainContent** classes are defined with `position: fixed; max-height: 100%; overflow: auto`.

When `position: fixed` is used, the resulting block is outside of the document flow.  Only the header and the skip paragraph are part of the document flow.

Based on the symptoms, my theory is that the **side** and **mainContent** divs get the window height (100%) and then the `top: 64px` adjusts the position pushing the bottom outside of the window.  This means the true height should be `100% - 64px`.

The CSS solution is to have a class that adds 64px to the bottom padding of the last object in Footer.shtml and parts/SidebarTail.shtml.

dsand47, 2021-07-07:
This has been fixed by changes to default.css in website PR #431.

closed 2021-07-12 by jerryg2003