JMRI/JMRI #9934: OSX Big Sur with M1 chip not working

pendelinopete, 2021-07-01:
I've just got a new MBP running OSX Big Sur 11.4 with an Apple M1 chip and when I launch Panel Pro, Decoder Pro or Sound Pro it fails to open with a Not enough permissions message. This is incorrect and have contacted Apple support who verified this and then suggested that I contact JMRI to see if the software is compatible with the M1 chip. So that's the issue.  Is it compatible with OSX 11.4 and Apple M1 chip?

dsand47, 2021-07-01:
> I've just got a new MBP running OSX Big Sur 11.4 with an Apple M1 chip and when I launch Panel Pro, Decoder Pro or Sound Pro it fails to open with a Not enough permissions message. This is incorrect [...]

The details for this issue are in issue #9197.  

In the meantime, you can install a x86_64 version of Java.  Rosetta will take care of Java and the third party supplied x86_64 code will get translated as well.

bobjacobsen, 2021-08-17:
Still working on it.

bobjacobsen, 2021-11-21:
JMRI 4.25.6 should basically work on M1 _except_ for JOAL Audio and libUSB support.

closed 2021-11-21 by bobjacobsen