JMRI/JMRI #9937: Listener location config is overwritten with defaults every time the panels are stored.

mrchuk, 2021-07-03:
I've been having unending trouble assigning VSD locatons to blocks and getting it to work properly, until I discovered that even though I've configured the default listener to 0,0 and set up the co-ordinates from there, JMRI is overwriting this with X=2, Y=1 every time I store the panels.

Reproducible by changing the default listener in the XML, and then storing a panel update (I used Blocks table, as I've been trying various combinations of x,y and making lots of updates).
I'm using 4.22.

Can JMRI please leave this alone once set?

klk32003, 2021-07-03:
Please check your VSD Preferences. It looks like you have configured the listener location to X=2, Y=1. VSD will, if started, use this setting.

mrchuk, 2021-07-04:
You are right--pilot error.
Issue cancelled.

closed 2021-07-04 by mrchuk