JMRI/JMRI #9958: mqtt singalhead topic is not correct

adarshkumarsingh83, 2021-07-12:
Issue Type: __Bug__

HI i am trying to add signal head and my panel pro is connected to the mosquitto mqtt and data is comming in 

/trains/track/turnout/3 CLOSED
/trains/track/turnout/2 THROWN
/trains/track/turnout/1 CLOSED

rather then 

/trains/track/singalhead/3 OFF
/trains/track/singalhead/2 ON
/trains/track/singalhead/1 OFF

PanelPro 4.22+R4c4384d74
Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 25.251-b08 mixed mode
Mac OS X 10.16 x86_64

adarshkumarsingh83, 2021-07-12:
for other like turnout and light and sensors are working very well only with signal head their is problem 
data is not pushed on correct topic from jmri panelpro 
its pushing in the turnout topic with signal number and throw/close value 
/trains/track/turnout/3 CLOSED
/trains/track/turnout/2 THROWN
/trains/track/turnout/1 CLOSED

 and expected is the singalhead topic with signal number and ON/OFF values 

/trains/track/singalhead/3 OFF
/trains/track/singalhead/2 ON
/trains/track/singalhead/1 OFF

danielb987, 2021-07-12:
Welcome to JMRI and thanks for your report!

Have you asked about this bug on the JMRI users list? There are much more people following that list than there are reading the Issues.
[Join JMRI users list](https://groups.io/g/jmriusers/join)
[Browse JMRI users list](https://groups.io/g/jmriusers/topics)

If you join the JMRI users list and post a message, it might take some days before any of the moderators have time to approve your message.

closed 2021-09-12 by stale[bot]