JMRI/JMRI #9966: V4.24 Turnout Oscillation

LeeThal54, 2021-07-14:
Issue Type: __Bug__

When running V4.24 turnout route script one of the turnouts does not complete and begins oscillating
OCCURS WITH v4.22 NOT v4.20
Equipment DCC Concept, NCE Power Cab, Java V8 build 291, WIN 10

PanelPro 4.24+Re18b309e8
Java HotSpot(TM) Client VM Oracle Corporation 25.291-b10 mixed mode
Windows 10 10.0 x86

danielb987, 2021-07-14:
Welcome to the JMRI project!

Is this bug report the same as in this message:
https://groups.io/g/jmriusers/message/192135

If so, I strongly suggest that you reply to the message from Billybob, showing him the script you are using and give him the console log.
https://groups.io/g/jmriusers/message/192136

We want to help, but in order for us to be able to help you, you need to provide more information, for example the script you are using. If you don't know what we are asking for, please ask and we will guide you. Also note that there are much more people reading the JMRI users list than it is people reading the issues, so you will get better help by answering the reply from Billybob than writing about the bug here.

LeeThal54, 2021-08-28:
I posted the scripts as requested and have had no response from anyone 
I’m at a loss to understand if there’s anything more I need to do 

Regards
Leigh 😎

danielb987, 2021-10-13:
Is this your message on the JMRI users list?
https://groups.io/g/jmriusers/message/192367

Does the oscillating stops if you doesn't load the script but only load the panel?

Does all the turnouts oscillate? If not, which turnouts oscillate?

Does the turnouts oscillate if you don't load the panel, but only load the script?

I'm pretty sure you will need to upload the panel file as well for us to be able to help you. You can upload the panel file as an attachment to a message in this issue. But you need to use the web site to do it, you cannot attach a file to an email since GitHub will remove all attachments from the email.

LeeThal54, 2021-10-14:
Yes this is my message on the JMRI user list - 192367
Since I've down graded to the previous version (4.22) which works OK, I can't answer your panel script question
It is only any turnout selected from the panel, not all of them.
I'll load the next version (again), and extract the panel file in the next few days and post it accordingly
BTW I've made no changes to the panel since V4.16

Regards

Leigh

LeeThal54, 2021-10-15:
Since the xml format is not suported here's the script
<?xml version="1.0" encoding="UTF-8"?>
<?xml-stylesheet href="/xml/XSLT/panelfile-4-19-2.xsl" type="text/xsl"?>
<layout-config xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="http://jmri.org/xml/schema/layout-4-19-2.xsd">
  <jmriversion>
    <major>4</major>
    <minor>20</minor>
    <test>0</test>
    <modifier />
  </jmriversion>
  <sensors class="jmri.jmrix.internal.configurexml.InternalSensorManagerXml">
    <defaultInitialState>unknown</defaultInitialState>
    <sensor inverted="false">
      <systemName>ISCLOCKRUNNING</systemName>
    </sensor>
  </sensors>
  <turnouts class="jmri.jmrix.nce.configurexml.NceTurnoutManagerXml">
    <operations automate="false">
      <operation name="NoFeedback" class="jmri.configurexml.turnoutoperations.NoFeedbackTurnoutOperationXml" interval="300" maxtries="2" />
      <operation name="Raw" class="jmri.configurexml.turnoutoperations.RawTurnoutOperationXml" interval="300" maxtries="1" />
      <operation name="Sensor" class="jmri.configurexml.turnoutoperations.SensorTurnoutOperationXml" interval="300" maxtries="3" />
    </operations>
    <defaultclosedspeed>Normal</defaultclosedspeed>
    <defaultthrownspeed>Restricted</defaultthrownspeed>
    <turnout feedback="DIRECT" inverted="true" automate="Off">
      <systemName>NT110</systemName>
      <userName>Upper Loop Entry</userName>
      <properties>
        <property>
          <key>WifiControllable</key>
          <value class="java.lang.Boolean">true</value>
        </property>
      </properties>
    </turnout>
    <turnout feedback="DIRECT" inverted="true" automate="Off">
      <systemName>NT120</systemName>
      <userName>South Inner Loop</userName>
      <properties>
        <property>
          <key>WifiControllable</key>
          <value class="java.lang.Boolean">true</value>
        </property>
      </properties>
    </turnout>
    <turnout feedback="DIRECT" inverted="false" automate="Off">
      <systemName>NT121</systemName>
      <userName>South Xing</userName>
      <properties>
        <property>
          <key>WifiControllable</key>
          <value class="java.lang.Boolean">true</value>
        </property>
      </properties>
    </turnout>
    <turnout feedback="DIRECT" inverted="true" automate="Off">
      <systemName>NT122</systemName>
      <userName>SE Xing</userName>
      <properties>
        <property>
          <key>WifiControllable</key>
          <value class="java.lang.Boolean">true</value>
        </property>
      </properties>
    </turnout>
    <turnout feedback="DIRECT" inverted="false" automate="Off">
      <systemName>NT123</systemName>
      <userName>NE Xing</userName>
      <properties>
        <property>
          <key>WifiControllable</key>
[... 851 lines cut ...]
                </operation>
              </filehistory>
            </operation>
            <operation>
              <type>Store</type>
              <date>Mon Apr 29 09:05:07 AWST 2019</date>
              <filename />
            </operation>
          </filehistory>
        </operation>
        <operation>
          <type>Store</type>
          <date>Mon Apr 29 13:24:38 AWST 2019</date>
          <filename />
        </operation>
      </filehistory>
    </operation>
    <operation>
      <type>Store</type>
      <date>Wed Dec 09 17:50:12 AWST 2020</date>
      <filename />
    </operation>
  </filehistory>
  
</layout-config>

dsand47, 2021-10-15:
You don't say which turnout is having the problem, but there is an issue in the yard routes.  

For example, look at route "IOYARD 02".

When "Yard 02" changes, it starts a script to set the yard ladder.  The route also "toggles" the "Yard 02" turnout.  This triggers the route again.

The routes need to be changed to trigger "On Closed" or "On Thrown".  The upper reference sets the turnout to the other state.

LeeThal54, 2021-10-16:
In answer to your 1st question - it wasn't any particular turnout.

I've been through and altered the route attributes to " On Closed" which affects appearance of the symbol on the panel and has stopped the loop effect which started this issue.

It still didn't explain the difference between V4.20 and V4.22 et al

Thanks for your help

Regards 
Leigh

danielb987, 2021-10-16:
> It still didn't explain the difference between V4.20 and V4.22 et al

I don't know. But it could be that a bug has been fixed between V4.20 and V4.22.

closed 2021-12-15 by stale[bot]