JMRI/JMRI #9973: JMRI XSLT pages have formatting and link issues

bobjacobsen, 2021-07-18:
JMRI constructs a few web pages from its XML files using XSLT. They're having some issues now, and the XSLT transforms probably need an update for the new web structure.

 - The [index](https://www.jmri.org/xml/XSLT/) has links to other pages made from decoder definitions, but those links are no longer working.

 - They're formatted for a sidebar (large whitespace to left), but have never had one.

I'm happy to do the XSLT edits if somebody let's me know what's missing in the HTML output.

dsand47, 2021-07-18:
The sidebar space can be suppressed by adding a class to the mainContent div.
```
<div id="mainContent" class="no-sidebar">
```

The attached file has the diff for two files to get the index pages to display properly and have the links work.

[diff.txt](https://github.com/JMRI/JMRI/files/6836492/diff.txt)

danielb987, 2021-07-19:
@bobjacobsen 

> * The [index](https://www.jmri.org/xml/XSLT/) has links to other pages made from decoder definitions, but those links are no longer working.

Is this problem fixed? If not, can you explain the problem? I haven't looked at that page before and I don't see the problem. Both the page itself and the pages it links to looks good to me, but I don't know how it's supposed to look like.

dsand47, 2021-07-19:
> @bobjacobsen
>
>
> Is this problem fixed? If not, can you explain the problem? I haven't looked at that page before and I don't see the problem. Both the page itself and the pages it links to looks good [...]

@danielb987 

The CSS changes for the move to HTML 5 caused the logo icon to overlay the top part of the content resulting in un-clickable links.

The diff file has the changes to fix this and other issues.  I don't know if there are other XSLT issues.

closed 2021-07-23 by danielb987