JMRI/JMRI #9993: Block Content display issue in Layout Editior

nschertz, 2021-07-25:
[session.log](https://github.com/JMRI/JMRI/files/6874617/session.log)
[systemInfo.txt](https://github.com/JMRI/JMRI/files/6874618/systemInfo.txt)
[profile.zip](https://github.com/JMRI/JMRI/files/6874619/profile.zip)

Issue Type: __Bug__

Profile: JMRI 4.24
Windows 10
Latest Java 8 release

Entered a block content display label from within the LE edit mode. Displays fine in black font with white background default. Changed the font color and background color with no change to the block content display label.  After adding a boarder the changed font color and background color shows up.  Apparently the change to add a boarder triggers a refresh, but the change of font color and background, by itself, does not.  Seems like a minor fix to trigger a refresh post font and/or background color change.

Thanks Norb

PanelPro 4.24+Re18b309e8
Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 25.291-b10 mixed mode
Windows 10 10.0 amd64

nschertz, 2021-07-25:
correct boarder to read border.. spell check gone bad. lol

geowar1, 2021-08-03:
It would be helpful to have the exact steps to reproduce.

dsand47, 2021-08-03:
George,

 1. Start with an empty PanelPro.
 2. Create a block.
 3. Create a new LE panel.
 4. Add a block contents icon.
 5. Do a store/stop
 6. Important --- Verify that there is no blockvalues.xml file.
 7. Do a start/load
 8. Use the context menu to set the block contents icon background color.
 9. -- no change to the block contents icon.
 10. Use the context menu to set the block contents border size to 1.
 11. -- the icon will show the color, but looks like just the border --
 12. Add a block value.
 13. -- now the block contents icon expands with the full background color.
 14. Remove the block value
 15. -- the icon shrinks to the empty size but has the full background color.

Dave Sand

nschertz, 2021-08-04:
Dave - Thanks for adding the step by step. NorbSent from my Galaxy

geowar1, 2021-08-04:
1. Start with an empty PanelPro.
 2. Create a block.
 **2a. Select "Tools" menu => "Tables" => "Blocks" item
 2b. "Blocks" window appears
 2c. Click "Add" button
 2d. "Add New Block" window appears
 2e. Enter system & user names and click the "Create" button
 2f.  You should see a new entry appear in the "Blocks" window's table
 2g. Click the "Cancel" button to close the "Add New Block" window**
 3. Create a new LE panel.
 4. Add a block contents icon.
 **4a. With the LE panel in edit mode select the "Block Contents" button.
 4b. From the menu next to the "Block Contents" button select the block you created in step 2.
 4c. Shift click in the middle of the LE edit pane to create a block contents icon.**
 5. Do a store/stop
 **5a. From the "File" menu select the "Store Panels" item
 5b. Enter a file name and location and then click the "Store" button
 5c. Quit PanelPro**
 6. Important --- Verify that there is no blockvalues.xml file [in the user files directory]
 7. Do a start/load
 8. Use the context menu to set the block contents icon background color.
 **8a. With the LE panel in edit mode drag-select a rectangle to find/hightlight the block contents icon.
 8b. Control- or right- click the block contents icon to bring up the contextual menu for the block contents icon.
 8c. Select the "Background Color" menu item
 8d. "Background Color" window should appear
 8e. Select a high contrast color (like red or bright green).
 8f. Click the "Ok" button
 8g. "Background Color" window should disappear**
 9. -- no change to the block contents icon.
 **9a. [Without a value it's autosize is zero; it doesn't think it has anything to redraw.]**
 10. Use the context menu to set the block contents border size to 1.
 11. -- the icon will show the color, but looks like just the border --
 12. Add a block value.
 13. -- now the block contents icon expands with the full background color.
 14. Remove the block value
 15. -- the icon shrinks to the empty size but has the full background color.

geowar1, 2021-08-04:
See PR #10024

closed 2021-08-23 by danielb987