JMRI/JMRI #10005: Duplicate Bundle Keys ( JmritOperationsTrainsBundle_en_GB )

icklesteve, 2021-07-28:
Current HOM Build failing due to Duplicate Bundle Keys.

It's fairly easy for these to slip in, is it worth investigating adding something like
http://rscbundlecheck.sourceforge.net/
to CI @danielb987  ?

```
[resourceCheck] Enabled checks: [duplicate key check, empty key check, messageformat check, unicode check, invalid char check]
[resourceCheck] duplicate key check: key buildReturnCarToStaging defined more than once (328:Returning wagon ({0}) to fiddle yard ({1}), no other tracks available) (/var/lib/jenkins/workspace/development/builds/java/src/jmri/jmrit/operations/trains/JmritOperationsTrainsBundle_en_GB.properties:309:Returning wagon ({0}) to fiddle yard ({1}), no other tracks available + 328:Returning wagon ({0}) to fiddle yard ({1}), no other tracks available)
[resourceCheck] duplicate key check: key buildStagingCanAcceptLoad defined more than once (396:Fiddle yard ({0}, {1}) can accept wagon''s load ({2})) (/var/lib/jenkins/workspace/development/builds/java/src/jmri/jmrit/operations/trains/JmritOperationsTrainsBundle_en_GB.properties:356:Fiddle yard ({0}, {1}) can accept wagon''s load ({2}) + 396:Fiddle yard ({0}, {1}) can accept wagon''s load ({2}))
```

danielb987, 2021-07-30:
@icklesteve 
Thanks for spotting this!

I have started looking at this in PR #10008. It might be good to add [ResourceCheck](http://rscbundlecheck.sourceforge.net/) as well, but there are a couple of things that check doesn't seem to address, which I think would be good to look at. The idea for #10008 is to check for:

* Duplicate keys in the same file
* A translated key that is not in the main bundle. The key exists in XNetSwingBundle_da.properties but not in XNetSwingBundle.properties. This could be because a key has been removed in the main bundle but not in the translated bundles.
* The same key exists in both a parent folder and a child folder. For example both in jmri/NamedBeanBundle.properties and in jmri/jmrit/Bundle.properties.

Maybe not all of these should be treated as errors, but I think it would be good to at least know the current state.

silverailscolo, 2021-07-30:
I support adding the check. Note that case #2 and case #4 can occur intentionally, where a translator sees need to use a different display string eg. in a dialog where the default (US) doesn't. For example where using the same string as a noun and a verb (object & action) is incorrect or when a translator wants to add some specifics.
Overriding a shared key in a package is also not common practice (better to create a new key?) but existing code might do that.

danielb987, 2021-08-02:
> Current HOM Build failing due to Duplicate Bundle Keys.

If it's important to fix this now, it would be good to do it in a separate PR. I'm working on the bundle check, but I have other things as well to work on, so I'm not sure when I will be able to finish this.

closed 2021-10-03 by DanielBoudreau