JMRI/JMRI #10007: Website: OperationsPro (and other) sub-pages are missing the side menu on the left

TrainzLuvr, 2021-07-30:
Tested using Chromium and Firefox browsers.

When loading https://www.jmri.org/help/en/html/apps/OperationsPro/index.shtml the side menu on the left appears and is functional.

Upon entering any of the sub-pages e.g. General Operations, Quck Start Guide, etc. the side menu on the left is not present.

It appears there are many other pages that also do not have the side menu, such as Layout Editor.

For the sake of UX, the expected behaviour would be for the side menu to be present.

danielb987, 2021-07-30:
@TrainzLuvr 
Welcome to JMRI and thanks for your report!

@dsand47 
Can you take a look at this?

dsand47, 2021-07-30:
The convention is that pages reachable using JMRI **Help &rArr; Window Help** are located in the **help/en/package** tree and do not have a sidebar.  As time and inclination permit, a CSS class is being added to allow the content to use the space that would normally have a sidebar.

Pages in the other **help/en** trees will normally have a sidebar.  There are some exceptions, such as the LogixNG reference chapters.  This was intentional to eliminate distractions.

TrainzLuvr, 2021-07-30:
Thanks for the prompt response!

I see what you mean. (actually I don't because I'm still using 4.22, and the Help is built-in) :)

Would it be worth while exploring an idea of identifying when a help page is accessed through the website vs from JMRI and then presenting the user with the menu in the former case?

JMRI could send something in the header or could probably be identified via an User-Agent on your server, and menu displayed or not.

dsand47, 2021-07-30:
As I mentioned, it is a **_convention_**.  There are pages in the **help/en/package** that do have sidebars.  The window help pages for the **LogixNG** tool have sidebars.  

The header, footer and sidebar are handled by SSI (server side includes).  When a page is displayed via the website, the includes are processed before sending the page to the browser.  JavaHelp (JmriHelp as of 4.25.1) does not process the includes so that content is not shown.

Adding a sidebar SSI to 269 pages is not a significant amount of work.  The big question is which sidebar?  A lot of window help pages have related content in the the **help/en/html** tree.  Extra work is required to make a sidebar work for both the **help/en/package** tree and the **help/en/html** tree.

TrainzLuvr, 2021-07-30:
I understand the issue with adding a specific side bar menu because it would be context sensitive, and would cause extra work.

If this is the convention, then no reason to change it. Getting to other pages via web is an extract click away, but not a problem.

Thanks.

closed 2021-07-30 by TrainzLuvr