JMRI/JMRI #10016: Windows get progressively smaller on Linux as JMRI is restarted

mrchuk, 2021-08-02:
I have noticed this issue with releases going back to at least 4.18. 
When I start JMRI, I have to resize all the windows because they keep shrinking every time I start JMRI. This affects at least the main window, Roster, Throttles, Thread Monitor and Script output.
Once I resize them, they will slowly keep shrinking again when I restart JMRI. Position is remembered correctly, but geometry is not. 
I've used xwininfo to capture the roster window geometry for two successive starts of JMRI:
Corners:  +114+149  -810+149  -810-1522  +114-1522
Corners:  +114+177  -816+177  -816-1521  +114-1521

I made no changes to the window sizes myself, but they are changing! Pretty sure it's not supposed to do this.

I am on Ubuntu Linux 18.04, JRE 1.8.0_301 and JMRI 4.24 but this has been happening for as far back as I can remember.

jerryg2003, 2021-08-02:
Do they keep shrinking on each successive restart, or does the size reach a “steady state”?  If this is due to some mathematical rounding issue, they may get to a steady state.

dsand47, 2021-08-02:
Look at profile/<random looking name>/user-interface.xml.

Do the window location and size values end with something other than .0?

```
<window xmlns="" class="jmri.jmrit.roster.swing.RosterFrame:Roster" height="488.0" locX="276.0" locY="55.0" width="846.0">
```

Pugwash1, 2021-08-02:
In the save restore code for sizing there is code that starts if os is linux... That is a bit dated and I think should be removed, what I don't know is if it's still needed in java 1.8 (latest) or is needed for specific Linux variants ( X vs Wayland). It's not needed in java 16 on fedora under Wayland, I do know that.
We should just remove the code, it cannot be worse than not having it in.
Steve G.

mrchuk, 2021-08-03:
Dave, they all end in .0 

Jerry, they may well reach steady state at some point but since they've all shrunk to unusable size before reaching it, I don't know.

Still looks like a bug and quacks like one :-D

Pugwash1, 2021-08-03:
@mrchuk 
Do they keep shrinking or do they keep getting narrower an deeper?
Is Ubuntu using X or Wayland?

mrchuk, 2021-08-04:
From inxi:
```
System:    Host: rodan Kernel: 4.15.0-151-generic x86_64 bits: 64
           Desktop: Gnome 3.28.4 Distro: Ubuntu 18.04.5 LTS
CPU:       Dual core Intel Core i7-6500U (-MT-MCP-) cache: 4096 KB
           clock speeds: max: 3100 MHz 1: 1032 MHz 2: 1993 MHz 3: 1535 MHz
           4: 2042 MHz
Graphics:  Card-1: Intel Skylake GT2 [HD Graphics 520]
           Card-2: Advanced Micro Devices [AMD/ATI] Mars [Radeon HD 8670A/8670M/8750M]
           Display Server: x11 (X.Org 1.19.6 ) drivers: i915,radeon
           Resolution: 1920x1080@60.02hz, 1920x1080@60.00hz
           OpenGL: renderer: Mesa DRI Intel HD Graphics 520 (SKL GT2)
           version: 4.6 Mesa 20.0.8
```

Also I have three measurements of the roster window from successive restarts. One value moving by +28 the other by -1
```
Corners:  +223+1436  -737+1436  -737-287  +223-287
Corners:  +223+1464  -743+1464  -743-286  +223-286
Corners:  +223+1492  -749+1492  -749-285  +223-285
```

mrchuk, 2021-08-06:
Guys, if you can point me to the code where this is all happening I can maybe have a swing at it myself. I'm not much of a java programmer myself but might be able to spot a problem in stuff that already exists.
Thanks1
Charles

icklesteve, 2021-08-07:
Hi @mrchuk 

See https://github.com/JMRI/JMRI/blob/master/java/src/jmri/util/JmriJFrame.java#L1015

Note 2014 comments from @DanielBoudreau 

Steve.

mrchuk, 2021-08-07:
Ouch! Nasty!

I'll set debugging on this and find out exactly what it's up to in my setup and report back.

Cheers, and thanks!

Charles

On 7/8/21 8:29 pm, Steve Young wrote:

Hi @mrchuk<https://github.com/mrchuk>

See https://github.com/JMRI/JMRI/blob/master/java/src/jmri/util/JmriJFrame.java#L1015

Note 2014 comments from @DanielBoudreau<https://github.com/DanielBoudreau>

Steve.

—

[Apulia Grove Oil logo]

Apulia Grove
40 Tranter Rd
Toolleen
Victoria 3551

M: 0409 258 471
www.apulia-grove.com.au <http://www.apulia-grove.com.au>

Pugwash1, 2021-08-09:
This applies to linux.
Big Problem 1: you cannot do a setSize(x,y) and then do a pack().. Hence all the messing with preferredsize thats in the code (which is only taken as a hint.)
Big Problem 2: resizeToFitScreen() does not use the getSize(), it uses this.getPreferredSize() so messes with the height and width again.
If you fix all this, there are still problems saving windows that are partially off screen or minimized.

bobjacobsen, 2021-08-09:
> you cannot do a setSize(x,y) and then do a pack().

Are you sure about this?

(Re)setting the size without a following `pack()` is explicitly cautioned against in The Swing Book.  A pack() before setting size would sized and arranged components within the manager’s constraints, but that sizing doesn’t get repeated when the overall size is set.  So things like viewports, etc, don’t resize for the new layout.

Pugwash1, 2021-08-09:
@bobjacobsen 
I am well aware of that statement and others that say never force the size of a window..But I could not find another way to make it work. If I set the size after the pack it works, before the pack it doesnt.
If someone can fix it using setPreferedSize thats fine, but no else has seemed to try and fix this on going festering sore.

Pugwash1, 2021-08-09:
@bobjacobsen 
Actually my previous attempt  (PR #6147) attacked it via the setpreferedsize(), but made a few of the screens, one of which was the roster, come up really tiny.

bobjacobsen, 2021-08-09:
I think the sequence
```
setSize(x,y)
pack()
setSize(x,y)
```
would be OK.  The first one would constrain the component layout, and then the second one makes sure the overall size is right. But I really don't understand why the second one is necessary:  `pack()` should never change the overall size of the component being packed.

Pugwash1, 2021-08-10:
@bobjacobsen 
If I go back to using setPrefferedSize() Ive discovered that after the pack() the frame is the correct size, but immediatle after the pack I get a componentresize event that has updated the size of the frame!!!!!

Pugwash1, 2021-08-10:
@bobjacobsen 
OK. If I use setPreferredSize() everywhere, and the do:
        super.setMaximumSize(super.getPreferredSize());
        super.setMinimumSize(super.getPreferredSize());
        super.pack();
        super.setMaximumSize(null);
        super.setMinimumSize(null);
It works.. The pack was returning before completing its thing. So its the pack adding a bit at the bottom, and taking a bit off the side. If I dont set the min and maxs back to null I cannot resize.
I dont know if its relevant or not as I only have a shadow of a memory about spacing during packs.

Pugwash1, 2021-09-25:
Fixed via #10081

closed 2021-09-25 by Pugwash1