JMRI/JMRI #10042: Editor dialog does not open when attempting to edit a roster entry in Decoder Pro

dball5000, 2021-08-14:
[profile.zip](https://github.com/JMRI/JMRI/files/6987182/profile.zip)
[session.log](https://github.com/JMRI/JMRI/files/6987183/session.log)
[systemInfo.txt](https://github.com/JMRI/JMRI/files/6987185/systemInfo.txt)

Issue Type: __Bug__

1. Open decoder pro
2. Right-click an existing roster entry and select Program

Expected: the editor dialog opens allowing me to make changes to the setting for the selected roster entry

Actual: the editor does not open and the program can now only be closed by killing the process in Task Manager. A window corresponding to the editor does appear in the Window menu and if I select that window, I do see a blank window - I cannot close this window though (I assume this is why the process must be killed in Task Manager) 

DecoderPro 4.24+Re18b309e8
Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 25.301-b09 mixed mode
Windows 10 10.0 amd64

dball5000, 2021-08-14:
One other thing I should mention. I am accessing the computer via remote desktop. I will try and  logon directly and see if this helps.

dball5000, 2021-08-14:
The direct logon made no difference

dball5000, 2021-08-14:
So got it working by renaming Basic.xml (in xml\programmers) to Basic.xml.xml. I assume my install has some how become slightly corrupted. Anywhere I can edit so that I can just use Basic.xml as I am supposed to?

KenC57, 2021-08-14:
Dball5000,

Instead you go into the Preferences for Roster. Select into the 'Programmer' tab. Then change the Format of
the programmer to something other than Basic. Save and exit JMRI. Undo the renaming you did for the Basic.xml.
Now you can go back into the Programmer tab and reselect Basic if you want that format.

Under some unknown case, the extra .xml gets stuck on the end of the entry it is looking for. Picking a
different one solves this.

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.com
www.syracusemodelrr.org

dball5000, 2021-08-15:
Thanks. That fixed the immediate problem. This happened after upgrading from 3.10 to the 4.24. Oddly, I always used the Comprehensive programmer in 3.10, not the Basic programmer that was set by default after the upgrade (though maybe there were no options in 3.10 - I never investigated because there was no need).