JMRI/JMRI #10047: Java exception when MQTT broker connection lost

jerryg2003, 2021-08-17:
After three retries to find the MQTT broker, JMRI 4.25.2 (and earlier) goes to a Java exception.  Is this what does JMRI do when it loses connection to other connection types?  Perhaps something more graceful... [In this example, I purposefully shut down the broker to see what would happen]

2021-08-17 02:05:08,961 mqtt.MqttAdapter                      WARN  - Lost MQTT broker connection... [MQTT Rec: ri9CB6D09A38D53db5210dM]
2021-08-17 02:05:08,961 mqtt.MqttAdapter                      INFO  - ...trying to reconnect [MQTT Rec: ri9CB6D09A38D53db5210dM]
2021-08-17 09:00:07,680 mqtt.MqttAdapter                      WARN  - Lost MQTT broker connection... [MQTT Rec: ri9CB6D09A38D53db5210dM]
2021-08-17 09:00:07,686 mqtt.MqttAdapter                      INFO  - ...trying to reconnect [MQTT Rec: ri9CB6D09A38D53db5210dM]
2021-08-17 09:00:45,260 mqtt.MqttAdapter                      WARN  - Lost MQTT broker connection... [MQTT Rec: ri9CB6D09A38D53db5210dM]
2021-08-17 09:00:45,260 mqtt.MqttAdapter                      INFO  - ...trying to reconnect [MQTT Rec: ri9CB6D09A38D53db5210dM]
2021-08-17 09:00:49,299 mqtt.MqttAdapter                      ERROR - Unable to reconnect [MQTT Rec: ri9CB6D09A38D53db5210dM]
Unable to connect to server (32103) - java.net.ConnectException: Connection refused: connect
	at org.eclipse.paho.client.mqttv3.internal.TCPNetworkModule.start(TCPNetworkModule.java:80)
	at org.eclipse.paho.client.mqttv3.internal.ClientComms$ConnectBG.run(ClientComms.java:724)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.net.ConnectException: Connection refused: connect
	at java.net.DualStackPlainSocketImpl.waitForConnect(Native Method)
	at java.net.DualStackPlainSocketImpl.socketConnect(Unknown Source)
	at java.net.AbstractPlainSocketImpl.doConnect(Unknown Source)
	at java.net.AbstractPlainSocketImpl.connectToAddress(Unknown Source)
	at java.net.AbstractPlainSocketImpl.connect(Unknown Source)
	at java.net.PlainSocketImpl.connect(Unknown Source)
	at java.net.SocksSocketImpl.connect(Unknown Source)
	at java.net.Socket.connect(Unknown Source)
	at org.eclipse.paho.client.mqttv3.internal.TCPNetworkModule.start(TCPNetworkModule.java:74)
	... 2 more

ear9mrn, 2021-11-09:
I am seeing the same issue. I have MQTT configured in connections.  I then start a script with the following.

` mqttAdapter = jmri.InstanceManager.getDefault( jmri.jmrix.mqtt.MqttSystemConnectionMemo ).getMqttAdapter()`

if i simply restart mosquitto I get the following and have to restart JMRI to get MQTT up and running again.

021-11-09 19:54:46,607 mqtt.MqttAdapter                      WARN  - Lost MQTT broker connection... [MQTT Rec: ri8888888887883ebbef16M]
2021-11-09 19:54:46,608 mqtt.MqttAdapter                      INFO  - ...trying to reconnect [MQTT Rec: ri8888888887883ebbef16M]
2021-11-09 19:54:46,627 mqtt.MqttAdapter                      ERROR - Unable to reconnect [MQTT Rec: ri8888888887883ebbef16M]
Unable to connect to server (32103) - java.net.ConnectException: Connection refused (Connection refused)
	at org.eclipse.paho.client.mqttv3.internal.TCPNetworkModule.start(TCPNetworkModule.java:80)
	at org.eclipse.paho.client.mqttv3.internal.ClientComms$ConnectBG.run(ClientComms.java:724)
	at java.base/java.lang.Thread.run(Thread.java:829)
Caused by: java.net.ConnectException: Connection refused (Connection refused)
	at java.base/java.net.PlainSocketImpl.socketConnect(Native Method)
	at java.base/java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:399)
	at java.base/java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:242)
	at java.base/java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:224)
	at java.base/java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.base/java.net.Socket.connect(Socket.java:609)
	at org.eclipse.paho.client.mqttv3.internal.TCPNetworkModule.start(TCPNetworkModule.java:74)
	... 2 more

running PanelPro 4.24+Rel81b309e8.

Other than that things seem to work fine.

Thanks.
Pete.

bobjacobsen, 2021-11-09:
There are two issues.

One shows a “ConcurrentModificationError”.  A proposed fix for that is in the works, and should be in a development release shortly.

The one below doesn’t have a proposed fix, because I haven’t been able to find a (proposed) problem.  The broker is dropping the connection, and I don’t know why.  

> On Nov 9, 2021, at 2:57 PM, ear9mrn ***@***.***> wrote:
> 
> 
> I am seeing the same issue. I have MQTT configured in connections. I then start a script with the following.
> 
> mqttAdapter = jmri.InstanceManager.getDefault( jmri.jmrix.mqtt.MqttSystemConnectionMemo ).getMqttAdapter()
> 
> if i simply restart mosquitto I get the following and have to restart JMRI to get MQTT up and running again.
> 
> 021-11-09 19:54:46,607 mqtt.MqttAdapter WARN - Lost MQTT broker connection... [MQTT Rec: ri8888888887883ebbef16M]
> 2021-11-09 19:54:46,608 mqtt.MqttAdapter INFO - ...trying to reconnect [MQTT Rec: ri8888888887883ebbef16M]
> 2021-11-09 19:54:46,627 mqtt.MqttAdapter ERROR - Unable to reconnect [MQTT Rec: ri8888888887883ebbef16M]
> Unable to connect to server (32103) - java.net.ConnectException: Connection refused (Connection refused)
> at org.eclipse.paho.client.mqttv3.internal.TCPNetworkModule.start(TCPNetworkModule.java:80)
> at org.eclipse.paho.client.mqttv3.internal.ClientComms$ConnectBG.run(ClientComms.java:724)
> at java.base/java.lang.Thread.run(Thread.java:829)
> Caused by: java.net.ConnectException: Connection refused (Connection refused)
> at java.base/java.net.PlainSocketImpl.socketConnect(Native Method)
> at java.base/java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:399)
> at java.base/java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:242)
> at java.base/java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:224)
> at java.base/java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
> at java.base/java.net.Socket.connect(Socket.java:609)
> at org.eclipse.paho.client.mqttv3.internal.TCPNetworkModule.start(TCPNetworkModule.java:74)
> ... 2 more
> 
> running PanelPro 4.24+Rel81b309e8.
> 
> Other than that things seem to work fine.

danielb987, 2021-11-09:
@bobjacobsen 
What happens if you:
1. Start the broker
2. Start JMRI and JMRI connects to the broker
3. Shut down the broker. Maybe hard (without the broker being able to tell JMRI what has happened)
4. Restart the broker
5. Will JMRI try to reconnect to the broker now? I think it should, independent of what has just happened. (But I haven't used MQTT so I might be wrong).

bobjacobsen, 2021-11-09:
JMRI can't do step 5, because it has no idea that step 4 has happened.

ear9mrn, 2021-11-09:
When I restart my broker JMRI does nothing has it has errored out (java error/crashed) and it is no longer trying to reconnect. It just needs to not crash when it loses connection to the broker and keep retrying to reconnect or listen until the port is alive again then try to connect.

danielb987, 2021-11-12:
See PR #10291.

closed 2022-01-11 by stale[bot]