JMRI/JMRI #10055: Exception thrown when help set to online and web page cannot be reached

pnbgit, 2021-08-22:
If the preferences are set to use the online help, and whilst offline the user selects help from the menu, an exception is thrown "Failed to connect to web page".  If the user does not have the system console open, the effect is that absolutely nothing happens when help is selected, it has ceased to do anything.

The same symptom would happen if, for any reason, the JMRI help website was down.

Would it be good to catch this exception, and then revert to local help whilst offline, perhaps with a popup message to that effect?

dsand47, 2021-08-22:
The changes have been implemented in PR #10058.

A dialog is displayed when the local help is being used after a web error.

[image]

closed 2021-08-23 by danielb987

pnbgit, 2021-09-01:
Great stuff - thanks for putting that fix in.