JMRI/JMRI #10083: V4.24 - Edit ConditionalNG loses existing Comment text when editing existing item

JIM-MOOMAW, 2021-09-03:
If A or E type line item having comment text is edited, that comment text is removed after edit OK button pressed.  Screenshots illustrate.

Workaround is to click-open the Comment and OK that before saving any code changes from edit screen.

[image]
[image]
[image]

danielb987, 2021-09-05:
Welcome to the JMRI project and thanks for your bug report!

The error is confirmed and I'm working on it.

danielb987, 2021-09-06:
The bug is now fixed by #10087 and will be resolved in the next test release JMRI 4.25.3.

closed 2021-09-06 by danielb987