JMRI/JMRI #10085: LogixNG Clipboard has stopped working, can't unjam

JIM-MOOMAW, 2021-09-04:
Within this data file I started using Clipboard for one-line copy-paste operations which was sorta helpful.  Then discovered I could copy-paste whole ranges of code which is hugely more helpful.   However after about a dozen large-code copy-paste operations the Clipboard has ceased functioning... won't Paste and doesn't seem to be copying code either.

To un-jam I tried using Notepad++ to remove the <Clipboard> section from the data file, but that didn't solve the problem... it still won't copy and paste anything.  So I assume cleanup is a bigger problem to restore operation.  The "Open Clipboard" contents doesn't appear to match anything in the <Clipboard> section, if that makes any difference.

Since PanelPro LogixNG doesn't lend itself to hand-coding, but requires endless tedious box checking, having the ability to copy and paste 90%-close code that can then be tweaked is a **huge** time saver as well as error reducer, so not having the Clipboard anymore for this unfinished data file is a big loss.  

I'll post the bug here, and ask for fix suggestions at JMRI groups.io.  

[image]

[image]

[image]

[clipboard contents in data file.txt](https://github.com/JMRI/JMRI/files/7109322/clipboard.contents.in.data.file.txt)

dsand47, 2021-09-04:
Any errors in the system console?  If so, use the Copy to clipboard button and paste here.

dsand47, 2021-09-04:
You should also include the xml data file (zipped or renamed with .txt) so that we have a test case to work with.

JIM-MOOMAW, 2021-09-04:
Actually I think the Console will be useful.  I just noticed I could use Copy/Paste on single lines at least, prior to seeing the error thrown in the Console, but once that happened the Clipboard was dead until quitting and starting a new PanelPro session.  Included XML file with TXT type.  

[Console for copy-paste A2 section of IQ-AUTO-0006.txt](https://github.com/JMRI/JMRI/files/7110249/Console.for.copy-paste.A2.section.of.IQ-AUTO-0006.txt)

[WMRC_Signal_Test_Bench v20 - dead Clipboard version.txt](https://github.com/JMRI/JMRI/files/7110250/WMRC_Signal_Test_Bench.v20.-.dead.Clipboard.version.txt)

JIM-MOOMAW, 2021-09-04:
OK, done.

dsand47, 2021-09-04:
We may also need a zip file of your custom icons and any other custom components.

dsand47, 2021-09-04:
That is an interesting problem.  I am able to re-create it.  

I suspect that the E10 socket name is the problem.  Maybe going to 2 digits is an issue.

A temporary work around is to use **Rename socket** from the context menu, such as XX.  Don't forget to change the formula.

JIM-MOOMAW, 2021-09-04:
I just sent this signal kit over to Jacobsen earlier today for inclusion in a future release... he filed them in GitHub someplace.  Their file paths here are set up for the local project folder the data file is run out of.  

[WPRR-1971 resources.zip](https://github.com/JMRI/JMRI/files/7110574/WPRR-1971.resources.zip)

I'm guessing these are the balance of the custom icons. 

[Custom Icons B.zip](https://github.com/JMRI/JMRI/files/7110596/Custom.Icons.B.zip)
-------------------------------------------------------------------

This little XML data file is for demo only and doesn't have the complete set of factors I usually use for clearing controlled signals.  The productional code will routinely have 12-18 evaluation terms in the digital equations.  I guess I'm saying I'll routinely have two-digit "E's" I'd like to copy and paste... especially in cases the code is complex and repetitive, so I hope that becomes feasible in the fixed-up version.

dsand47, 2021-09-04:
I have a copy of the WP PR so signals are not an issue.

The En names are the default, but you can rename them.  Using more descriptive names may help with complex formulas.

Another approach is to create modules.  

An unrelated question:  Have you looked at Tools => CTC?  https://www.jmri.org/help/en/package/jmri/jmrit/ctc/CTC.shtml

JIM-MOOMAW, 2021-09-05:
Your point about using other LogixNG features, like Modules, is a good one.  For this little project I thot I’d use techniques I know from Logix, and once I get those working as a baseline in the new NG environment I’ll come back and look at alternate approaches.  LogixNG is a pretty big lift itself trying to figure out.  I find myself wishing there were more coding examples to help me understand how to begin applying the new stuff.  Perhaps I should copy this traditional approach XML and then hack it up with alternate solutions.  

 

--JimM

JIM-MOOMAW, 2021-09-05:
Data Point:  Modified the same IQC with the E10 value that threw error, with new socket names with single digits.  That worked, and no Console errors.  Then created another socket to paste this too.  That worked.  Code runs.  So this is clearly a workaround.  And as far as I'm concerned far better for setting up and reading the equations... great suggestion.  

[image]

JIM-MOOMAW, 2021-09-05:
Data Point:  Attempted to copy whole IQC at "A" and Console threw error.  Console error and IQC attached.  This may be a bogus thing to do with the Clipboard, but seems a logical way to replicate an IQC I want to modify for another use.   This code should be the same as you have in the XML I sent.   

After restarting PanelPro I tried copy/paste of A6 and that worked fine... no Console errors.  

[image]

[image]

JIM-MOOMAW, 2021-09-05:
Data Point:  After fresh restart tried to copy/paste A1 of same IQC, and Console threw error.  None of these E expressions is two digit, so something else is going on.   Including copy of XML v24... some changes since the v20 I sent.  

[Console for copy A1 section of IQC-AUTO-0021.txt](https://github.com/JMRI/JMRI/files/7111129/Console.for.copy.A1.section.of.IQC-AUTO-0021.txt)

[WMRC_Signal_Test_Bench v24.TXT](https://github.com/JMRI/JMRI/files/7111132/WMRC_Signal_Test_Bench.v24.TXT)

dsand47, 2021-09-05:
@danielb987 

You are going to have to take this one.  I have run out of ideas.

danielb987, 2021-09-05:
> You are going to have to take this one. I have run out of ideas.

I'm working on it.

danielb987, 2021-09-06:
@JIM-MOOMAW 
Thanks for the bug report! I think I have found the cause and I have a solution for it in PR #10089.

JIM-MOOMAW, 2021-09-06:
Prior to a fixed release it appears I can avoid Clipboard fails by avoiding out of sequence numbers… a workaround.  I’ll keep an eye out for those.   Tx.

JIM-MOOMAW, 2021-09-06:
Changing names of the terms is a brilliant idea… I will do that certainly for the reasons you give.  That will also give a clue as to Clipboard behaviors.

 

I’ll check out the CTC link… looks interesting.    Thanks for the help!

danielb987, 2021-09-07:
PR #10089 is now merged and will be included in the next test release JMRI 4.25.3.