JMRI/JMRI #10090: Turnout Table checkboxes not rendering correctly.

dheap, 2021-09-06:
Seeing this problem on all my Macs with both 4.25.2 and Head of Master.
[image]

 Sierra & Yosemite.

Dave

KenC57, 2021-09-06:
Dave,

Check the setting in the Edit->Preferences->Display, changing it may give a clue to the problem. I always use
the 'Metal' setting and it seems to avoid some issues or make things clearer for me. At least at some point in
time I found that setting made things look right for me and I've stuck with it since.

dsand47, 2021-09-06:
The turnout and sensor table options were changed to use the jmri.util.swing.TriStateJCheckBox.

This was in PR #9410 and included in JMRI 4.21.5.

dheap, 2021-09-10:
It seems no suitable icons are available for my versions of Mac OS X Java LAF  with Java 1.8.

`INFO  - DecoderPro version 4.25.3ish+heap+20210910T1043Z+R7db3fd67c5 starts under Java 1.8.0_301 on Mac OS X x86_64 v10.12.6 at Fri Sep 10 20:43:52 AEST 2021 [main]`

There's a sort-of workaround for Nimbus... Tried it for Mac OS X LAF but no joy...

icklesteve, 2021-09-23:
Hi @dheap 

I've found an improved fix for Nimbus just by setting the selected icon during initialisation, eg.

`setSelectedIcon(icon);`

This, however still requires a valid icon file.
I'm unable to repreduce this so could you please post the output from

`log.warn(javax.swing.UIManager.getDefaults().toString());`

Thanks, Steve.

dheap, 2021-09-24:
@icklesteve 
```
2021-09-24 12:47:28,421 swing.TriStateJCheckBox               WARN  - laf is: Mac OS X Defaults = {Table.ancestorInputMap.RightToLeft=javax.swing.plaf.InputMapUIResource@6ed9f5d4, TextField.highlight=javax.swing.plaf.ColorUIResource[r=255,g=255,b=255], Button.border=com.apple.laf.AquaButtonBorder$Dynamic@51b279c9, EditorPane.caretForeground=javax.swing.plaf.ColorUIResource[r=0,g=0,b=0], Spinner.editorBorderPainted=true, TextPane.focusInputMap=javax.swing.plaf.InputMapUIResource@51016767, ScrollBar.focusInputMap=javax.swing.plaf.InputMapUIResource@731f8236, TabbedPane.foreground=javax.swing.plaf.ColorUIResource[r=0,g=0,b=0], PrintingDialog.contentProgress.textAndMnemonic=Printed page {0}..., PopupMenu.foreground=javax.swing.plaf.ColorUIResource[r=0,g=0,b=0], CheckBoxMenuItem.foreground=javax.swing.plaf.ColorUIResource[r=0,g=0,b=0], AbstractDocument.styleChange.textAndMnemonic=style change, TabbedPane.selectedLabelShift=-1, activeCaptionText=javax.swing.plaf.ColorUIResource[r=0,g=0,b=0], FileChooser.upFolderIcon=javax.swing.plaf.IconUIResource@1da51a35, TextArea.selectionForeground=com.apple.laf.AquaImageFactory$SystemColorProxy[r=0,g=0,b=0], Table.selectionBackground=com.apple.laf.AquaImageFactory$SystemColorProxy[r=9,g=80,b=208], FileChooser.by.textAndMnemonic=Name, com.apple.laf.AquaSpinnerUI=class com.apple.laf.AquaSpinnerUI, TreeUI=com.apple.laf.AquaTreeUI, FormattedTextField.background=javax.swing.plaf.ColorUIResource[r=255,g=255,b=255], TextField.font=com.apple.laf.AquaFonts$DerivedUIResourceFont[family=Lucida Grande,name=Lucida Grande,style=plain,size=13], Focus.color=com.apple.laf.AquaImageFactory$SystemColorProxy[r=48,g=131,b=251], AbstractDocument.addition.textAndMnemonic=addition, List.sourceListBackgroundPainter=com.apple.laf.AquaListUI$ComponentPainter@6ee787cc, TextArea.inactiveBackground=javax.swing.plaf.ColorUIResource[r=255,g=255,b=255], Spinner.font=com.apple.laf.AquaFonts$DerivedUIResourceFont[family=Lucida Grande,name=Lucida Grande,style=plain,size=13], class com.apple.laf.AquaSpinnerUI=public static javax.swing.plaf.ComponentUI com.apple.laf.AquaSpinnerUI.createUI(javax.swing.JComponent), CheckBoxMenuItem.font=com.apple.laf.AquaFonts$DerivedUIResourceFont[family=Lucida Grande,name=Lucida Grande,style=plain,size=14], Table.focusCellForeground=com.apple.laf.AquaImageFactory$SystemColorProxy[r=164,g=205,b=255], PasswordField.border=com.apple.laf.AquaTextFieldBorder@4516af24, PopupMenu.selectionForeground=javax.swing.plaf.ColorUIResource[r=255,g=255,b=255], FileChooser.useSystemExtensionHiding=false, List.sourceListFocusedSelectionBackgroundPainter=com.apple.laf.AquaListUI$ComponentPainter@231a37e5, Button.textShiftOffset=0, Menu.selectionForeground=javax.swing.plaf.ColorUIResource[r=255,g=255,b=255], FormView.resetButton.textAndMnemonic=Reset, ProgressBarUI=com.apple.laf.AquaProgressBarUI, ComboBox.buttonHighlight=javax.swing.plaf.ColorUIResource[r=255,g=255,b=255], ColorChooser.hsbRed.textAndMnemonic=R, FileChooser.updateButton.textAndMnemonic=Update, TabbedPane.highlight=javax.swing.plaf.ColorUIResource[r=255,g=255,b=255], TextField.border=com.apple.laf.AquaTextFieldBorder@4516af24, _SecurityDecisionIcon=javax.swing.plaf.IconUIResource@63a3a7e, TextField.background=javax.swing.plaf.ColorUIResource[r=255,g=255,b=255], Spinner.actionMap=javax.swing.plaf.ActionMapUIResource@55d9e340, CheckBoxMenuItem.disabledForeground=javax.swing.plaf.ColorUIResource[r=128,g=128,b=128], TextField.darkShadow=javax.swing.plaf.ColorUIResource[r=0,g=0,b=0], List.oddRowBackgroundPainter=com.apple.laf.AquaListUI$ComponentPainter@18f606e2, ColorChooser.font=javax.swing.plaf.FontUIResource[family=Dialog,name=Dialog,style=plain,size=12], PopupMenu.actionMap=javax.swing.plaf.basic.LazyActionMap@4b9e255, OptionPane.warningIcon=javax.swing.plaf.IconUIResource@64d2d351, RadioButtonMenuItem.foreground=javax.swing.plaf.ColorUIResource[r=0,g=0,b=0], List.background=javax.swing.plaf.ColorUIResource[r=255,g=255,b=255], CheckBox.textIconGap=4, RadioButton.textIconGap=4, OptionPaneUI=com.apple.laf.AquaOptionPaneUI, TextArea.inactiveForeground=javax.swing.plaf.ColorUIResource[r=128,g=128,b=128], OptionPane.minimumSize=javax.swing.plaf.DimensionUIResource[width=262,height=90], TableHeaderUI=com.apple.laf.AquaTableHeaderUI, PasswordField.selectionForeground=com.apple.laf.AquaImageFactory$SystemColorProxy[r=0,g=0,b=0], com.apple.laf.AquaToolBarUI=class com.apple.laf.AquaToolBarUI, MenuBar.selectionBackground=com.apple.laf.AquaImageFactory$SystemColorProxy[r=48,g=131,b=251], PopupMenu.background=javax.swing.plaf.ColorUIResource[r=255,g=255,b=255], TableHeader.ancestorInputMap=javax.swing.plaf.InputMapUIResource@4f9a3314, Tree.foreground=javax.swing.plaf.ColorUIResource[r=0,g=0,b=0], PasswordField.capsLockIconColor=javax.swing.plaf.ColorUIResource[r=0,g=0,b=0], html.pendingImage=sun.swing.ImageIconUIResource@176d53b2, PasswordField.caretBlinkRate=500, FileChooser.directoryOpenButtonMnemonic=0, class com.apple.laf.AquaToolBarUI=public static javax.swing.plaf.ComponentUI com.apple.laf.AquaToolBarUI.createUI(javax.swing.JComponent), Tree.ancestorInputMap=javax.swing.plaf.InputMapUIResource@1b68b9a4, DesktopIcon.border=javax.swing.plaf.BorderUIResource$CompoundBorderUIResource@4856fbab, InternalFrame.optionDialogBackground=com.apple.laf.AquaImageFactory$SystemColorProxy[r=238,g=238,b=238], MenuItem.margin=javax.swing.plaf.InsetsUIResource[top=0,left=0,bottom=0,right=0], TextField.selectionBackground=com.apple. [... cut ...]  com.apple.laf.AquaTableUI, Menu.acceleratorSelectionForeground=javax.swing.plaf.ColorUIResource[r=0,g=0,b=0], TextField.shadow=javax.swing.plaf.ColorUIResource[r=142,g=142,b=142], controlDkShadow=javax.swing.plaf.ColorUIResource[r=0,g=0,b=0], Table.dropLineColor=javax.swing.plaf.ColorUIResource[r=142,g=142,b=142], MenuBar.font=com.apple.laf.AquaFonts$DerivedUIResourceFont[family=Lucida Grande,name=Lucida Grande,style=plain,size=14], Button.defaultButtonFollowsFocus=true} [AWT-EventQueue-0]

```

dheap, 2021-09-24:
@icklesteve with context:
```
JMRI.runtime-library-selection:
arch.lib.path /Users/heap/jmri/lib/macosx/x86_64:/Users/heap/jmri/lib/macosx
Launch normally (no debugger support)
2021-09-24 13:18:32,883 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2021-09-24 13:18:32,888 util.Log4JUtil                        INFO  - This log is appended to file: /Users/heap/Library/Preferences/JMRI/log/messages.log [main]
2021-09-24 13:18:32,888 util.Log4JUtil                        INFO  - This log is stored in file: /Users/heap/Library/Preferences/JMRI/log/session.log [main]
2021-09-24 13:18:32,896 apps.AppsBase                         INFO  - DecoderPro version 4.25.4ish+heap+20210924T0247Z+R986a784 starts under Java 1.8.0_301 on Mac OS X x86_64 v10.12.6 at Fri Sep 24 13:18:32 AEST 2021 [main]
2021-09-24 13:18:39,469 gui3.Apps3                            INFO  - Starting with profile NCE_Simulator.3ee5400a [main]
2021-09-24 13:18:39,659 node.NodeIdentity                     INFO  - Using jmri-aR5vnbySGhNiaaBm-FPR4L-3ee5400a as the JMRI storage identity for profile id 3ee5400a [AWT-EventQueue-0]
2021-09-24 13:18:39,928 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "NCE" [main]
2021-09-24 13:18:40,258 simulator.SimulatorAdapter            INFO  - NCE Simulator Started [Nce Simulator]
2021-09-24 13:18:40,260 nce.NceConnectionStatus               INFO  - NCE EPROM revision = 6.2.2 [AWT-EventQueue-0]
2021-09-24 13:18:40,364 nce.NceConnectionStatus               INFO  - Connected to NCE command station [AWT-EventQueue-0]
2021-09-24 13:18:43,418 startup.PerformFileModel              INFO  - Loading file /Ours/Our Documents/Dave/Dropbox/JMRI/Test Light-It Panel.xml [main]
2021-09-24 13:18:46,303 swing.TriStateJCheckBox               WARN  - laf is: Mac OS X Defaults = {TextField.highlight=javax.swing.plaf.ColorUIResource[r=255,g=255,b=255], Button.border=com.apple.laf.AquaButtonBorder$Dynamic@51b279c9, EditorPane.caretForeground=javax.swing.plaf.ColorUIResource[r=0,g=0,b=0], Spinner.editorBorderPainted=true, ScrollBar.focusInputMap=javax.swing.plaf.InputMapUIResource@731f8236, TabbedPane.foreground=javax.swing.plaf.ColorUIResource[r=0,g=0,b=0], PrintingDialog.contentProgress.textAndMnemonic=Printed page {0}..., PopupMenu.foreground=javax.swing.plaf.ColorUIResource[r=0,g=0,b=0], CheckBoxMenuItem.foreground=javax.swing.plaf.ColorUIResource[r=0,g=0,b=0], AbstractDocument.styleChange.textAndMnemonic=style change, TabbedPane.selectedLabelShift=-1, activeCaptionText=javax.swing.plaf.ColorUIResource[r=0,g=0,b=0], FileChooser.upFolderIcon=javax.swing.plaf.IconUIResource@1da51a35, TextArea.selectionForeground=com.apple.laf.AquaImageFactory$SystemColorProxy[r=0,g=0,b=0], Table.selectionBackground=com.apple.laf.AquaImageFactory$SystemColorProxy[r=9,g=80,b=208], FileChooser.by.textAndMnemonic=Name, com.apple.laf.AquaSpinnerUI=class com.apple.laf.AquaSpinnerUI, TreeUI=com.apple.laf.AquaTreeUI, FormattedTextField.background=javax.swing.plaf.ColorUIResource[r=255,g=255,b=255], TextField.font=com.apple.laf.AquaFonts$DerivedUIResourceFont[family=Lucida Grande,name=Lucida Grande,style=plain,size=13], Focus.color=com.apple.laf.AquaImageFactory$SystemColorProxy[r=48,g=131,b=251], AbstractDocument.addition.textAndMnemonic=addition, TextArea.inactiveBackground=javax.swing.plaf.ColorUIResource[r=255,g=255,b=255], Spinner.font=com.apple.laf.AquaFonts$DerivedUIResourceFont[family=Lucida Grande,name=Lucida Grande,style=plain,size=13], class com.apple.laf.AquaSpinnerUI=public static javax.swing.plaf.ComponentUI com.apple.laf.AquaSpinnerUI.createUI(javax.swing.JComponent), CheckBoxMenuItem.font=com.apple.laf.AquaFonts$DerivedUIResourceFont[family=Lucida Grande,name=Lucida Grande,style=plain,size=14], Table.focusCellForeground=com.apple.laf.AquaImageFactory$SystemColorProxy[r=164,g=205,b=255], PasswordField.border=com.apple.laf.AquaTextFieldBorder@4516af24, PopupMenu.selectionForeground=javax.swing.plaf.ColorUIResource[r=255,g=255,b=255], FileChooser.useSystemExtensionHiding=false, Button.textShiftOffset=0, Menu.selectionForeground=javax.swing.plaf.ColorUIResource[r=255,g=255,b=255], FormView.resetButton.textAndMnemonic=Reset, ProgressBarUI=com.apple.laf.AquaProgressBarUI, ComboBox.buttonHighlight=javax.swing.plaf.ColorUIResource[r=255,g=255,b=255], ColorChooser.hsbRed.textAndMnemonic=R, FileChooser.updateButton.textAndMnemonic=Update, TabbedPane.highlight=javax.swing.plaf.ColorUIResource[r=255,g=255,b=255], TextField.border=com.apple.laf.AquaTextFieldBorder@4516af24, TextField.background=javax.swing.plaf.ColorUIResource[r=255,g=255,b=255], CheckBoxMenuItem.disabledForeground=javax.swing.plaf.ColorUIResource[r=128,g=128,b=128], TextField.darkShadow=javax.swing.plaf.ColorUIResource[r=0,g=0,b=0], ColorChooser.font=javax.swing.plaf.FontUIResource[family=Dialog,name=Dialog,style=plain,size=12], PopupMenu.actionMap=javax.swing.plaf.basic.LazyActionMap@4b9e255, OptionPane.warningIcon=javax.swing.plaf.IconUIResource@64d2d351, RadioButtonMenuItem.foreground=javax.swing.plaf.ColorUIResource[r=0,g=0,b=0], List.background=javax.swing.plaf.ColorUIResource[r=255,g=255,b=255], CheckBox.textIconGap=4, RadioButton.textIconGap=4, OptionPaneUI=com.apple.laf.AquaOptionPaneUI, TextArea.inactiveForeground=javax.swing.plaf.ColorUIResource[r=128,g=128,b=128], OptionPane.minimumSize=javax.swing.plaf.DimensionUIResource[width=262,height=90], TableHeaderUI=com.apple.laf.AquaTableHeaderUI, PasswordField.selectionForegrou [... cut ...] af.AquaMenuItemUI=class com.apple.laf.AquaMenuItemUI, Table.dropLineColor=javax.swing.plaf.ColorUIResource[r=142,g=142,b=142], RadioButton.font=com.apple.laf.AquaFonts$DerivedUIResourceFont[family=Lucida Grande,name=Lucida Grande,style=plain,size=13], MenuBar.font=com.apple.laf.AquaFonts$DerivedUIResourceFont[family=Lucida Grande,name=Lucida Grande,style=plain,size=14], Spinner.actionMap=javax.swing.plaf.ActionMapUIResource@6b1b021c, Button.defaultButtonFollowsFocus=true} [AWT-EventQueue-0]

```

icklesteve, 2021-10-21:
Other JCheckBox Icons implement UIResource so I've a feeling this could be the main issue?

bobjacobsen, 2022-05-13:
This looks OK to me on macOS 12.3.1 Java 11.0.14.1

Is this perhaps fixed?

icklesteve, 2022-05-15:
closing pending further reports.

closed 2022-05-15 by icklesteve