JMRI/JMRI #10096: Enable Polling at Startup - still not working - more details

RandyBartelt, 2021-09-08:
I'm just starting to use JMRI. I've set up a CMRI serial node with 2 sensors and 2 turnouts as a POC. My JMRI version is 4.24+Rel8b309e8.  I'm using Windows 10, my input and output pins run off an Arduino Mega, a Mega sensor shield, and a PCA9685.
The problem is twofold:
1) The Enable polling at startup always remains checked, no matter how I set it.  And the following holds true whether I've just saved preferences after attempting to check or uncheck the setting.
2) After each restart of PanelPro, the status continuously flips between INIT an POLLING, one cycle per second, and none of the sensors or turnouts actually get polled. I have to Halt Polling and then Resume Polling in order to get the pins to work. I may have to do this Halt/Resume procedure several times until I no longer see the INIT status, just POLLING. Then the input and output pins start working as expected.

RandyBartelt, 2021-09-08:
I should add that my USB port baud rate is 19,200.

KenC57, 2021-09-09:
Randy,

If the node isn't responding, dropping the baud rate will help it. Even at 9600, a local layout with three
SMINI and two USIC nodes gets each node serviced about 20 times per second which seems to work fine. Given how
most Arduino nodes work, the slower speed may help. They seldom do decent interrupt handling unless special
coding is used.

RandyBartelt, 2021-09-10:
Thank you very much for your suggestion.  I, too, thought that might be the problem.  Unfortunately, after switching everything on that port to 9600 baud, the behavior I described remains the same.  
No need to reply any further, since I've been urged to take up the issue on JMRI group.

Thanks,
Randy Bartelt

closed 2021-11-09 by stale[bot]

ianmidd, 2022-06-11:
Linked to ISCLOCKRUNNING. I clicked this to Inactive and Polling started

RandyBartelt, 2022-06-15:
My startup script has been setting all my sensors to INACTIVE.  This has unintentionally also set ISCLOCKRUNNING to INACTIVE.  I hadn't put the two together, but I now realize why I have not had the polling problem any more.
Thanks.
This issue can be closed.  However, I am curious why the default setting for ISCLOCKRUNNING, which is ACTIVE, would cause the polling problem, and if so, why is the default not INACTIVE.

ianmidd, 2022-06-15:
Good question. 

Regards,

Ian
+44 7768 215798