JMRI/JMRI #10114: Cannot add Light Controller of type Turnout Status when Light and Turnout share the number

pkmiec, 2021-09-16:
Hi! 

I using SPROG connection for DCC and C/MRI connection for Sensors / Lights (via arduino). I have a Turnout on the SPROG connection ST1 and a Light CL1 on C/MRI connection. When I try to create a Turnout Status Light Controller for CL1 with ST1, the dialog shows "Error: Turnout could not be provided..." (i.e. LightWarn2). See attached,

[image]

I believe there are two improvements that could be made here (both in the ,
  - The wrong error is shown. The actual error is LightWarn3 (https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/beantable/light/AddEditSingleLightControlFrame.java#L528) but it is overridden by LightWarn2 (https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/beantable/light/AddEditSingleLightControlFrame.java#L557). 
  - I am not sure whether this situation should trigger LightWarn3. Even though, the Turnout and Light share the same number, they are attached to different connections and thus I believe that should work.

Note, that attempting to create a Light SL1 (on the SPROG connection) correctly fails with a message that a Turnout on the connection with the same number already exists.

I am fairly new to JMRI, so I apologize if I said something naive.

danielb987, 2021-09-16:
@pkmiec 
Welcome to the JMRI project and thanks for your report!

Have you asked about this on the JMRI users mailing list or the JMRI developers mailing list? There are more people reading these lists than listening to new Issues, so you might get help quicker if you do.

If you ask on the JMRI users mailing list or the JMRI developers mailing list, please add a link to this issue.

KenC57, 2021-09-16:
Paul K,

You are correct that the code around 528 is ignoring if the system prefix for the light and turnout are the
same. It should skip this is unless it has matched the system prefixes of the light and turnout.

pkmiec, 2021-09-17:
Thanks for the quick responses. Sounds like this is something to be fixed. I can attempt to do that. I've read https://www.jmri.org/help/en/html/doc/Technical/gitdeveloper.shtml and it seems pretty straight forward. I'll post a PR when ready.

jerryg2003, 2021-09-17:
Paul - 

Welcome!  What version are you using?  I'm on 4.25.2 and was successful in setting up light controller as you specified (using both SPROG and C/MRI simulator connections).  Would there be different behavior if I wasn't using the simulators?

Looking more at the code you linked to, I wonder if the issue of light and turnout have the same system number is in fact only an issue if both are in C/MRI  (in which case, JMRI would have given you a warning when trying to set up a light with the same number as a turnout - which would not typically be allowed if both using C/MRI.   If this is true, then more of the code is called into question.  Your example has turnout under "S" and light under "C" so shouldn't be any conflict anyway.

[image]

KenC57, 2021-09-17:
One thought occurred to me. Should this tool even care about lights and turnouts being possibly in conflict?
The create bean logic for a given system connection should enforce the restriction of lights vs turnouts if
the restriction exists. Yes that restriction exists for CMRI but my point is what area of code should care.
Only the create should care, this tool should simply look at the two things it is working with.

This tool, which is simply making a connection between two beans (things) it should only make sure they aren't
the identical bean. And the one has to be of type light and the other of type turnout.

pkmiec, 2021-09-18:
@jerryg2003 Hi. I first noticed the problem with 2.24, but then I downloaded the source and reproduced / debugged it on effectively the latest master. I am using simulator for both sprog and cmri at the moment, but looking at the code in question, it does not seem that would matter. 

@KenC57 It makes sense to me the enforcement at the creation time.

pkmiec, 2021-09-18:
@danielb987 Hi, no I haven't hopped onto the mailing list. I will make sure to check in the future.

icklesteve, 2021-09-25:
Hey @pkmiec 

I've had a review of the code within https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/beantable/light/AddEditSingleLightControlFrame.java

Unfortunately it's not just a case of changing warning messages.
The multi-connection issue also applies to the other Light Controllers, eg. 2 Sensor Light Controllers, Timed Sensor etc.

There's also a decent amount of legacy code from when a normal JComboBox was used for Bean selection, not NamedBeanComboBox.
A few lines of code earlier, the NamedBeanComboBoxes are supplying a real Bean, I can't see a reason for the multiple checks to ensure it's a real Bean.

I've tested #10139  with multiple CAN connections and now allows the same number for different connections.

Steve.

icklesteve, 2021-09-30:
@pkmiec Could you please check functionality now that the update has been merged?  Thanks, Steve.

jerryg2003, 2021-09-30:
Happy to test.  Is there a jar file I can download?  Unfortunately, I do not have my own development environment.
_________________________________
***@***.***

danielb987, 2021-09-30:
I think this has the changes:
https://builds.jmri.org/jenkins/job/development/job/packages/lastStableBuild/

pkmiec, 2021-09-30:
@icklesteve Yes, I'll test it tonight. Thanks!

jerryg2003, 2021-09-30:
I created turnout ST1 and sensor CL1 with light controller with no errors, as expected.  When I tried to add turnout CT1, I got a warning that light 1 already existed, as expected.  So functionality works as expected in these cases. 

However, I did note that after creating a C/MRI Light, when I went to create another light, the “default” showed up as it was when I entered JMRI for one of my other connections, and did not change to C/MRI as I would have expected.  The “change of default based on last object created” does work correctly for turnouts.

Jerry

pkmiec, 2021-10-01:
Works great for me! Thank you.

danielb987, 2021-11-01:
@pkmiec 
Is this issue completed? If so, can it be closed?
It's easier for us to see which Issues that needs attention if the Issues that are completed are closed.

pkmiec, 2021-11-01:
Yes. Sorry.

closed 2021-11-01 by pkmiec