JMRI/JMRI #10130: Unrecoverable error encountered

forty-ford, 2021-09-23:
10 year user...
Currently using 4.25.3 with Java 1.8.0_301 (en)

Core dump...  Using Logix with LogixNG open

This has happened twice now within a few minutes.  I made a screen shot of the ERROR message, but it's a WINDOWS .rtf file and I don't know how to convert it so I can send it to you.

Thanks,
Donn K. Mueller

dsand47, 2021-09-23:
The contents of JMRI system console would be much better.  If it is not open, go to `Help => System Console`.  Use the `Copy to clipboard` button and paste the content in a reply.

forty-ford, 2021-09-23:
I cannot open the system console once the error occurs.  The JMRI program windows are all inoperable while the message is displayed and the screens dump (disappear) as soon as I close the unrecoverable error.  When I try reloading, it says it's already running.  I shut down, reboot and restart JMRI and the problem reoccurs.

I am trying the system on another computer, to ensure I'm not having a computer problem. I'll let you know what's happening.

dsand47, 2021-09-23:
The system console content is also located at C:\Users\<username>\JMRI\log\session.log.  After re-booting but before starting JMRI again, upload the file.

forty-ford, 2021-09-23:
Here's the log...

Can you explain the ''delayed sensor action" warnings?  I have sensors that I delay the action so the process will work.  If I don't delay the sensor action the the sensor is changed by another process after the original process has executed...
Thank you,
Donn

dsand47, 2021-09-23:
It appears that you did not upload a file.  You need to do that on GitHub.

forty-ford, 2021-09-23:
Okay...

forty-ford, 2021-09-24:
Dave,
I placed the messages.log in:
https://groups.io/g/jmriusers/files/ProblemsBeingWorkedOn/Donn Mueller - Unrecoverable Error
I also encountered the same problem on my laptop, so I'm sure that there is nothing wrong with my computer.
Thanks,Donn

KenC57, 2021-09-24:
Forty-four,

You should upload the panel file being loaded. It seems that something is getting into a loop and consuming
all of the memory and then crashing. We likely can find what is wrong in that file.

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.com
www.syracusemodelrr.org

dsand47, 2021-09-24:
The first problem is that you are loading 3 separate xml data files.
- R-E-RR-v-4-25-03-15.xml
- 099-5-Master Control Panel.xml
- 000-17-Summit Main Line.xml

The Java stack overflow occurs after loading the third file.  This occurs when a Logix is repeatedly trying to set a signal head appearance.  It could also be a Logix A triggers Logix B triggers Logix A triggers Logix B loop.

Loading more than one data file rarely works.  If the three files are totally independent, then it might work.  

The `timer already active on request to start delayed sensor action` warning says that a Logix (either the same one or another one) is requesting a delayed action for a sensor while a previous delayed action for the same sensor still has time left.

forty-ford, 2021-09-24:
Ken,
I've placed the file in the same location.
Thank you,Donn

dsand47, 2021-09-24:
You need to upload a zip file of your custom icons and any other custom components.

dsand47, 2021-09-24:
Here is the Logix for the first timer error in the system console log.

```
  IX-LS-X-DCO-LT50-LS9-LEDC1  C1   
    [ ]  R1  IF  Sensor "IS-PANEL-POWER-TGL" state is "Sensor Active"   
    [x]  R2    AND Sensor "LS9" state is "Sensor Active"   
    [x]  R3    AND Turnout "LT50" state is "Turnout Thrown"   
             THEN   
               When Triggered True, Set Sensor, "IS-LS-X-DCO-LT50-LS9-LED" to Active   
               When Triggered False, Delayed Set Sensor, "IS-LS-X-DCO-LT50-LS9-LED" to Inactive, after 3 seconds.   
```
If the Logix evaluates to false when either LS9 or LT50 changes state, the sensor starts a 3 second time.  If another change occurs and the result is still false and 3 seconds has not passed, an attempt to set the sensor inactive after 3 seconds will occur.  This results in the warning.

This occurs because the Logix is using the "when triggered" option instead of the "when changed" option.  If the "when changed" option was used, the first false evaluation would result in the 3 second delay.  Subsequent false evaluations would have no impact until a true evaluation has occurred.  

The warnings can still occur.  For example, if R1 and R3 are true and R2 is changing several times in 3 seconds.

forty-ford, 2021-09-25:
Hi Dave,
Last night I tried to find a loop, but could find none.  I have 12 to 16 reversing loops, for which I have planned.
I did find an error  in signal elements (a missing sensor) between two other sensors LS112 & LS9,  when I added the sensor the error messages stopped.   I worked for an hour after fixing the error without an error.
I will let you know what happens.
Thank you for the "when triggered" vs. "when changed" clarification!!!
I'm assuming that --

     <conditional systemName="IX-IH1C1" userName="Overide Green Signal #1 to Flashing Yellow WHEN LS11 ACTIVE &amp; LT28 CLOSED OR LT95 THROWN &amp; IH1 GREEN" antecedent="(R1 and R2 and R3 and R5) OR (R1 and R2 and R4 and R5)" logicType="3" triggerOnChange="no">
      <systemName>IX-IH1C1</systemName>
      <userName>Overide Green Signal #1 to Flashing Yellow WHEN LS11 ACTIVE &amp; LT28 CLOSED OR LT95 THROWN &amp; IH1 GREEN</userName>
      <conditionalStateVariable operator="4" negated="no" type="1" systemName="IS-PANEL-POWER-LED" dataString="" num1="0" num2="0" triggersCalc="no" />
      <conditionalStateVariable operator="1" negated="no" type="1" systemName="LS11" dataString="" num1="0" num2="0" triggersCalc="yes" />
      <conditionalStateVariable operator="1" negated="no" type="4" systemName="LT28" dataString="" num1="0" num2="0" triggersCalc="yes" />
      <conditionalStateVariable operator="1" negated="no" type="3" systemName="LT95" dataString="" num1="0" num2="0" triggersCalc="yes" />
      <conditionalStateVariable operator="1" negated="no" type="13" systemName="IH1" dataString="Green" num1="0" num2="0" triggersCalc="yes" />
      <conditionalAction option="1" type="3" systemName="IH1" data="8" delay="0" string="Flashing Yellow" />
    </conditional>

-- should be --

    <conditional systemName="IX-IH1C1" userName="Overide Green Signal #1 to Flashing Yellow WHEN LS11 ACTIVE &amp; LT28 CLOSED OR LT95 THROWN &amp; IH1 GREEN" antecedent="(R1 and R2 and R3 and R5) OR (R1 and R2 and R4 and R5)" logicType="3" triggerOnChange="yes">
      <systemName>IX-IH1C1</systemName>
      <userName>Overide Green Signal #1 to Flashing Yellow WHEN LS11 ACTIVE &amp; LT28 CLOSED OR LT95 THROWN &amp; IH1 GREEN</userName>
      <conditionalStateVariable operator="4" negated="no" type="1" systemName="IS-PANEL-POWER-LED" dataString="" num1="0" num2="0" triggersCalc="no" />
      <conditionalStateVariable operator="1" negated="no" type="1" systemName="LS11" dataString="" num1="0" num2="0" triggersCalc="yes" />
      <conditionalStateVariable operator="1" negated="no" type="4" systemName="LT28" dataString="" num1="0" num2="0" triggersCalc="yes" />
      <conditionalStateVariable operator="1" negated="no" type="3" systemName="LT95" dataString="" num1="0" num2="0" triggersCalc="yes" />
      <conditionalStateVariable operator="1" negated="no" type="13" systemName="IH1" dataString="Green" num1="0" num2="0" triggersCalc="yes" />
      <conditionalAction option="1" type="3" systemName="IH1" data="8" delay="0" string="Flashing Yellow" />
    </conditional>

???
I'm also guessing that "triggersCalc="yes" should remain and not be changed to "no"?  Is that correct? Please advise...  Is there an on line explanation of the difference between using the "when triggered" option and the "when changed" option?

Thank you and Ken for all your time and help.
Donn

P.S.    I worked today, so this is the first chance to read your email and reply.   Thanks again.

dsand47, 2021-09-25:
The "execute" options are described at https://www.jmri.org/help/en/html/tools/Logix.shtml#thepolicy

In the sample I posted, IX-LS-X-DCO-LT50-LS9-LEDC1, R1 is unchecked.  This means its changes have no impact until either R2 or R3 trigger a calculation.  The "execute" option will determine if the actions should be executed.

closed 2021-11-24 by stale[bot]