JMRI/JMRI #10163: Max OS X, Big Sur 11.5.2

Eric-Ashley, 2021-10-03:
I am having trouble with the installation and initial use of JMRI.  I am using SPROG III as my command station along an Arduino Mega as a sensor hub.  I have the latest release of JMRI install.  When I click on any of the main components of JMRI (Panel Pro, etc), the icon appears in the tray, but does not open.  I have Java 1.8 installed.  Any help is appreciated...[image]

dsand47, 2021-10-03:
Use Finder and go to /Users/<username>/Library/Preferences/JMRI/log.  Attach the launcher.log and session.log files.

Eric-Ashley, 2021-10-04:
Thanks for the immediate reply!

 

The Arduino is recognized on USB14100 and the SPROG is recognized on USB14200.  It does not give a baud rate…

 

The Arduino is working with Sensors, Servos, and lighting via CRMI…

 

I am anxious to begin.  I have a plan drawn in AnyRail and plan to import to JMRI to automate and take full advantage of all that JMRI has to offer…

 

I just need a little help over the initial installation.  

 

The instructions indicate I do not need any additional USB drivers.

 

I only have MAC Platforms.  I would rather not purchase a PC based laptop just to make this work.  However, LokProgrammer only works in a windows environment.  I hope to use an emulator (Virtual Machine) to make it work.

 

Again, thanks for the immediate response and help!

 

Eric

danielb987, 2021-10-04:
@Eric-Ashley 
Welcome to the JMRI project!

Please read the response from @dsand47 carefully:
`Use Finder and go to /Users//Library/Preferences/JMRI/log. Attach the launcher.log and session.log files.`

We need the content of these two files to understand what's wrong. Without that, we cannot help you.

Eric-Ashley, 2021-10-04:
The two log files were attached to my response…

Eric Ashley
Mobile: (757) 544-7224

danielb987, 2021-10-04:
It seems that GitHub doesn't accept email attachments. I think you need to go to the GitHub issue and add the log file to a reply with the web browser.
https://github.com/JMRI/JMRI/issues/10163

Eric-Ashley, 2021-10-04:
Thank you!

I am working at the fire station.  I will not be able to upload the documents until Tuesday night…

Eric Ashley
Mobile: (757) 544-7224

Eric-Ashley, 2021-10-05:
I am sorry for the delay.  Here is the Information...
[launcher.log](https://github.com/JMRI/JMRI/files/7287219/launcher.log)
[session.log](https://github.com/JMRI/JMRI/files/7287221/session.log)
[image]
[image]

dsand47, 2021-10-05:
Did you intend to close this issue?

According to the session.log file you are running 4.22, not 4.24.  I suspect the session.log is from a previous run.

The actual problem is that your install of JMRI is "quarantined".  This is indicated by content of the launcher.log file.

This is most likely caused by how you installed 4.24.  Try re-installing using these steps:
- Download the DMG from jmri.org.
- Open the DMG.
- Drag the JMRI icon to the Applications icon.
- Reply **Replace**

Eric-Ashley, 2021-10-05:
No joy...  The same thing happens.  I have used privacy to allow the app(s), but they do not ask for permission when opened.  Still the same issue, no Panel Pro, No Decoder Pro, No Sound Pro...

Eric-Ashley, 2021-10-05:
I see in groups.io that you have a new script for Panel Pro, but I cannot locate the script in groups.io.  Is there a new script for each panel?

Eric-Ashley, 2021-10-06:
Sorry, Still no joy...  I have disabled gatekeeper.  I am not sure what else to day.  In the above, I ask some additional questions...

dsand47, 2021-10-06:
Open a terminal window:
```
cd /Applications/JMRI
xatter -l PanelPro.app
```
-l is dash el

The response will look like the following:

com.apple.quarantine: 01c3;60e7c241;Safari\x20Technology\x20Preview;56634A0D-335D-4022-9A0B-A0E1837CF44E

The important part is at the beginning.  It needs to 01 not 00.

Eric-Ashley, 2021-10-06:
This command worked…

 

sudo xattr -r -d com.apple.quarantine /applications/JMRI/PanelPro.app

 

Eric

Eric-Ashley, 2021-10-06:
Clearing the quarantine extended attribute from downloaded applications

November 20, 2012rtroutonLeave a commentGo to comments

When I’m packaging an application for deployment, I normally download the latest version from the vendor. If it’s a drag-and-drop install, where the application is a self-contained bundle, it will get the com.apple.quarantine extended attribute associated with it.

The quarantine attribute is added by the OS so that it can ask for user confirmation the first time the downloaded program is run. Gatekeeper in Lion and Mountain Lion also uses the com.apple.quarantine attribute to determine if an application should be checked (and blocked if needed.)

To avoid issues with both unwanted messages and Gatekeeper, I normally clear this attribute as part of my packaging. Here’s how to check to see if your application has the com.apple.quarantine attribute associated with it:

1. Open Terminal

2. Run the following command:

xattr /path/to/MyApp.app

If the com.apple.quarantine attribute is associated with the application, you should see the following output:

computername:Applications username$ xattr /path/to/MyApp.app
com.apple.quarantine
computername:Applications username$

Note: Other attributes may also be listed, with com.apple.FinderInfo being a common one.

To remove the quarantine attribute, you would then run the following command:

sudo xattr -r -d com.apple.quarantine /path/to/MyApp.app

That will recursively remove the com.apple.quarantine attribute from the application. The -r option will allow the quarantine attribute of all files inside the application to be selected, while the -d option causes the given attribute name (and associated value) to be removed.

dsand47, 2021-10-06:
The "second command" was describing what the `-l` looks like.  Sometimes it get confused with "I" (eye) depending on the font.

I forgot to mention that you should post the output from `xatter -l PanelPro.app`.  There is no problem with the quarantine flag.  The important part is the '01c3" vs. '00c3".   This shows whether the app is notarized.  If it is not notarized, then it runs in a "sandbox".  JMRI is not designed to work in a sandbox.

All of the recent JMRI downloads are signed and notarized.  Since your downloads tried to run in a sandbox implies there is something else going on in your system.

The likely candidates are running from Documents, Downloads or Desktop, modifying the JMRI install by adding custom components, using an anti-virus program, or other unknowns.

Removing the quarantine is a brute force solution that hides your actual problem.