JMRI/JMRI #10165: Decoder pro vs loksound v5 xl

Spud44, 2021-10-04:
Ive been using decoder pro for some time now, but recently i've been given a loksound V5 to fit.

For some reason i cant get the decoder to identify as a V5xl.   i've tried paged and direct bit, and the responses i get are that it identifies as a Roco 10745,  an ESU switchpilot and ESU Lokpilot, but not as what it should be!

the decoder is brand new, and it works ( default address etc..). I've even tried a factory reset, all to no avail.

i've even installed JMRI onto  a newer PC and the latest version on windows 10 ( just in case the decoder descriptions were a bit screwy on my old XP machine!)

Hoping someone can tell me whats going on!!!  All i want to do is edit the funtion table and turn the volume down !

 

many thanks.

danielb987, 2021-10-04:
@Spud44 
Welcome to the JMRI project!

@dheap 
Is this something you can help with?

dheap, 2021-10-05:
@Spud44 @danielb987 
> For some reason i cant get the decoder to identify as a V5xl. i've tried paged and direct bit, and the responses i get are that it identifies as a Roco 10745, an ESU switchpilot and ESU Lokpilot, but not as what it should be!

All LokSound decoders work best with Direct Mode and will return 151 for CV8 and 255 for CV7. Then a large value for ProductId.

we need the line from the log that looks like this:
`... decoderdefn.IdentifyDecoder           INFO  - Decoder returns mfgID:123;modelID:123;productID:-1 [...]`

danielb987, 2021-11-01:
@Spud44 
Is this issue resolved? If so, can it be closed?
It's easier for us to see which Issues that needs attention if the Issues that are completed are closed.

If you still have the problem, please provide the line from the log that looks like this:
`... decoderdefn.IdentifyDecoder INFO - Decoder returns mfgID:123;modelID:123;productID:-1 [...]`

closed 2021-12-31 by stale[bot]