JMRI/JMRI #10166: AAR Graphical Symbols for CTC Panels

Mark-W-L, 2021-10-04:
Issue Type: __Feature Request__

Some CTC panels use the AAR semaphore graphical symbols for signals - even if the signals themselves are now color lights. Attached are a suitable set of symbols to be added to the resources/icons directory. The AAR smashboard symbols are also included.

PanelPro 4.24+Re18b309e8
Java HotSpot(TM) Client VM Oracle Corporation 25.301-b09 mixed mode
Windows 10 10.0 x86

[AARGraphicalSymbols.zip](https://github.com/JMRI/JMRI/files/7279811/AARGraphicalSymbols.zip)

bobjacobsen, 2021-10-04:
In PR #10167 

Thank you!

Mark-W-L, 2021-10-05:
Reference is: American Railway Signaling Principles and Practices Chapter II. By the Signal Section, A.A.R. Revised August 1946. Figure 1 Graphical Symbols Wayside Signal Operating Characteristics ARA Sig.Sec. 1660A

Mark-W-L, 2021-10-06:
Here are the associated .bmp files
[AAR_1946_Graphical_Symbols_BMPs.zip](https://github.com/JMRI/JMRI/files/7293386/AAR_1946_Graphical_Symbols_BMPs.zip)

danielb987, 2021-11-01:
@Mark-W-L 
Is this issue completed? If so, can it be closed?
It's easier for us to see which Issues that needs attention if the Issues that are completed are closed.

Mark-W-L, 2021-11-01:
bobjacobsen noted above that these icons have been merged into the 4.25.4 build.  Issue can be closed.

closed 2021-11-01 by Mark-W-L