JMRI/JMRI #10168: Common CI error: File D:\a\JMRI\JMRI\temp\Scripts\EMRA OPs Power Cycle.py not found

danielb987, 2021-10-04:
A common error on Windows CI lately is:

`LoadAndStoreTest>LoadAndStoreTestBase.tearDown:390 Unexpected ERROR or higher messages emitted: File D:\a\JMRI\JMRI\temp\Scripts\EMRA OPs Power Cycle.py not found.`

Anyone who understands what the problem might be?

dsand47, 2021-10-05:
The script reference is in a Logix Conditional in`test/jmri/jmrit/display/layoutEditor/load/PU-MonP-Signal-Panel-4-19-6.xml`

It shows up two times:
`      <conditionalAction option="1" type="16" systemName=" " data="-1" delay="0" string="preference:Scripts/EMRA OPs Power Cycle.py" />`

Apparently the Logix is randomly executed by the LoadAnd Store test.

Maybe under certain conditions the load stays around long enough to finish initializing before being stopped.

dsand47, 2021-10-09:
If the JMRI fast clock time is between 00:00 and 00:01 or 12:00 and 12:01, the IX-OPS Logix will attempt to load the script.

PUlvestad, 2021-10-18:
I'm confused how these files are being referenced.
The script (EMRA OPs Power Cycle.py) is one I made and use at my club (Edmonton Model Railroad Assoc.) triggered by Logix.
PU-MonP-Signal-Panel-4-19-6.xml seems to reference my clubs panel )not the same name).

Bob J was given a copy of the panel at one time when he was testing changes to layout editor.