JMRI/JMRI #10170: Cannot Connect Computer and PR3

ewarhol, 2021-10-05:
Issue Type: __Bug__

I'm having so much difficulty getting my computer and Digitrax PR3 to work together. I keep getting Com port in use error messages. I've tried unplugging and plugging the USB cable back in, using different ports (Com 3 and Com 5). Tried restarting, power off, back on. Sometimes the power to the Chief system needs to be on, sometimes off. I'm just getting tired of spending 30-60 minutes of playing around trying to get this to work. I'm sure there's something simple I'm missing, but I'm out of ideas and looking for help.

DecoderPro 4.25.3+R96551bed89
Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 25.301-b09 mixed mode
Windows 10 10.0 amd64

ewarhol, 2021-10-05:
Just learned of DigilPL on the Digitrax website. I followed those directions and the PR3 connected on the first try. Will do more testing and get back to this post. Fingers crossed this fixed the issue.

danielb987, 2021-10-05:
@ewarhol 
Welcome to the JMRI project!

If you have more problems, I recommend that you ask on the JMRI users mailing list. There are more people reading that list than listening to new Issues, so you might get help quicker if you do.