JMRI/JMRI #10204: Add throttle button to roster development suggestion

POWEN462, 2021-10-18:
It would be very handy to be able to add a column to the roster table which has a button to launch a throttle for that particular loco.

danielb987, 2021-10-19:
@POWEN462 
Welcome to the JMRI project!
Thanks for your suggestion.

KenC57, 2021-10-19:
The current roster has that as a right click option menu (on Windows) and the current selected line and throttle button in the lower right corner of the roster.

POWEN462, 2021-10-19:
> The current roster has that as a right click option menu (on Windows) and the current selected line and throttle button in the lower right corner of the roster.

Yes agreed those exist, this would be an extra feature, one click on the roster line rather than either right click and select new throttle or select the loco and then click on the new throttle.

closed 2021-12-18 by stale[bot]