JMRI/JMRI #10242: Java Exception for Unlit Signals

Mark-W-L, 2021-10-30:
Issue Type: __Bug__

Many of the signal definitions do not include an Unlit signal. However, the Add/Edit Signal Mast dialog allows the 'This Mast can be unlit:' check box to be selected even if there is no unlit definition. If a signal is then  created using the DCC Signal Mast Decoder as the Mast Driver with the check box ticked a Java Exception occurs and an error box is displayed. The signal is still created although obviously it does not have an unlit aspect.

PanelPro 4.24+Re18b309e8
Java HotSpot(TM) Client VM Oracle Corporation 25.311-b11 mixed mode
Windows 10 10.0 x86

[session.log](https://github.com/JMRI/JMRI/files/7447544/session.log)
[systemInfo.txt](https://github.com/JMRI/JMRI/files/7447548/systemInfo.txt)

[Dialog-Box.txt](https://github.com/JMRI/JMRI/files/7447552/Dialog-Box.txt)

Mark-W-L, 2021-10-30:
I was trying to create a signal that would be dark until a train entered the block and would go dark again when the train left the succeeding block. It was only when I saw the exception that I remembered there was no Unlit signal defined.

dsand47, 2021-10-31:
Significant changes to Unlit were recently made and are included in the 4.25.4 test release.

Mark-W-L, 2021-12-17:
Tested in 4.26, now working correctly. Exception not raised and no error box displayed.

closed 2021-12-17 by Mark-W-L