JMRI/JMRI #10260: Clean up JmriHelp

dsand47, 2021-11-04:
The JmriHelp system, which replaced JavaHelp, uses stub files to redirect Window and General help requests from JMRI to the appropriate html file if the default local help file preference is active.  The stub file redirects to ../index.html#<help-reference_fragment> which uses Javascript to redirect to help-reference#fragment.  Java cannot directly cause the default browser to load a local file and redirect to a fragment.

The current implementation uses the JmriHelp_??TOC.xml and JmriHelp_??Index.xml files along with scanning the shtml help files for `id="names"` to build the stub files.  This results in 4600+ stub files.

Most of the of the JMRI links are created using  `JmriJFrame.addHelpMenu("help-reference_fragment");`.  The help name is the the help path and name with an optional `_fragment`.   These are used to build the Help menu entries.  There are 312 direct calls via JmriJFrame plus a small number of indirect JmriJFrame calls. There can also be direct calls to `HelpUtil.displayHelpRef(String)`.

This means we need less than 400 stub files for each language (en and fr).

Is there a tool or easy way to scan the JMRI source tree for the help references?

dsand47, 2021-11-05:
I found an example that user Files.walk().

bobjacobsen, 2021-11-05:
I did a regex search for a `"package.`:
```
% grep -r '"html\.' java/src |wc
       8      28     992
% grep -r '"package\.' java/src |wc
     420    1662   60275
```

You can then extract the first string from those:
```
% grep -r '"html\.' java/src | awk -F '"' '{print $2}'      
html.tools.TimeTable
html.tools.TimeTable
html.apps.DispatcherPro.index
html.apps.SoundPro.SoundPro
html.apps.DecoderPro.index
html.apps.DecoderPro.index
html.apps.PanelPro.index
html.apps.PanelPro.index
(base) jake@Jakes-MacBook-Pro-2 JMRI % grep -r '"package\.' java/src | awk -F '"' '{print $2}' |head
package.
package.jmri.jmrit.simpleclock.SimpleClockFrame
package.jmri.jmrit.signalling.AddEditSignalingLogic
package.jmri.jmrit.signalling.SignallingSourceFrame
package.jmri.jmrit.MemoryFrameAction
package.jmri.jmrit.beantable.LightTable
package.jmri.jmrit.beantable.LightAddEdit
package.jmri.jmrit.beantable.SignalMastRepeater
package.jmri.jmrit.beantable.SignalMastAddEdit
package.jmri.jmrit.beantable.SignalMastAddEdit
...
```

There are going to be some special cases, like that first reference to "package." above. It's in JFrame itself:
```
    public String getHelpTarget() {
        return "package." + this.getClass().getName();
    }
```
This might be the indirect call you mentioned?

dsand47, 2021-11-05:
I have decided to eliminate the pre-built stub files.

When `HelpUtil.displayHelpRef` is called and the default local help location preference is active, a `stub.html` file is created and used for the browser call.

Since this requires read/write access, the stub file is located in `settings:jmrihelp`.  

The `ant buildhelp` process will no longer build the stub files.  It is still used to create the webtoc.shtml, webindex.shtml and local/index.html files.

closed 2021-11-06 by dsand47