JMRI/JMRI #10266: Product ID

MarcFoum, 2021-11-06:
Writing up a definition file for Bachmann Econami SoundValue decoders and hitting a glitch.
The decoders have CV253 value of 4 and CV256 value of 4 which would make  High bit / Low bit  of 68 dec   which is what I used as a Product ID. 

Unfortunately New Loco  shows all of the decoders that are Cv7 = 70  and  not using (I believe)  the valid Product ID. 

I wonder if the fact it is a 2 decimal value is what is throwing it off track.. Just about all Product ID (Soundtraxx) are three decimal that I have seen.

Marc

danielb987, 2021-11-06:
@MarcFoum 
Are you on the JMRI Developers list? If so, I suggest that you write about this there. It seems more developers are listening on the JMRI mailing lists than reading the Issues on GitHub.

MarcFoum, 2021-11-06:
No I am not and if I can not get the answer from here I will drop the subject. They do not need the likes of me in there.  
The user is on JMRI 4.25.1  and I question if the code was damaged. I have no such decoder to test with.

I wonder what code and were that  code handling this is in JMRI.

dheap, 2021-11-06:
Marc,

> On 6 Nov 2021, at 1:24 pm, MarcFoum ***@***.***> wrote:
> 
> The decoders have CV253 value of 4 and CV256 value of 4 which would make High bit / Low bit of 68 dec which is what I used as a Product ID.

That gives a ProductID of 1,028 (= CV253 times 256 plus CV256).

You also need to do a Recreate Decoder Index because the ProductID etc. are cached in "decoderIndex.xml".

Dave in Australia

dheap, 2021-11-06:
Marc,

> On 6 Nov 2021, at 2:15 pm, MarcFoum ***@***.***> wrote:
> 
> The user is on JMRI 4.25.1 and I question if the code was damaged. I have no such decoder to test with.
> 
See my other email regarding calculating the correct ProductID value and rebuilding "decoderIndex.xml".
> I wonder what code and were that code handling this is in JMRI.
> 
The JavaDoc is at:
<https://www.jmri.org/JavaDoc/doc/jmri/jmrit/decoderdefn/IdentifyDecoder.html>

The code is at:
<https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/decoderdefn/IdentifyDecoder.java>

Dave in Australia

MarcFoum, 2021-11-06:
I found my answer just before getting the Email from Dave. By studying other definitions and the answer was 404 = 1028 . and not 44 - 68   It must be fix  and I sent the new (corrected) def to the user in question. Thank you

MarcFoum, 2021-11-06:
Closing  , issue resolved.

closed 2021-11-06 by MarcFoum