JMRI/JMRI #10281: Enhancement Question

level2programmer, 2021-11-09:
I may be missing how to do this, forgive me if I am, but would like to be able to create a train that leaves from Location A, go to Location B (fill/empty loads) then return to Location A.  We have live loading and wold like to take a train to a coal mine, fill the cars and then return to the same starting location.

danielb987, 2021-11-09:
@level2programmer 
Welcome to the JMRI project!

Have you asked about this on the JMRI users mailing list? There are more people reading that list than listening to new Issues, so you might get help quicker if you do.

danielb987, 2021-12-24:
@level2programmer 
I think [Dispatcher](https://www.jmri.org/help/en/package/jmri/jmrit/dispatcher/Dispatcher.shtml) is what you are looking for.

closed 2022-02-07 by danielb987