JMRI/JMRI #10287: C/MRI SMINI explanatory text addition proposed

rrgibbons, 2021-11-10:
Proposed text to be added into the ‘Creating an SMINI Node Definition’. Located at: 

https://www.jmri.org/help/en/html/hardware/cmri/CMRI.shtml#L391

I suggest inserting the following text just below the bulleted list in the indicated section. 
------------------------
“When setting up a SMINI in JMRI, be aware that JMRI doesn’t care that it sees three nominal ‘IN Cards’ and 6 ‘OUT Cards’ on a SMINI. JMRI will just present 24 input bits and 48 output bits for you to use for that SMINI. The SMINI nominal ‘Card 2’, Ports A, B, and C, with INPUT bits 0 through 7 for each port gets simplified into INPUT bits 1 through 24. The JMRI Bit Assignments window will automatically show the 24 INPUT pins on the SMINI as CSXX. If you have 2 or more SMINI’s, the INPUT pin names will reflect the additional nodes. The SMINI nominal ‘Card 0’ and ‘Card 1’, each with their own Ports A, B, and C, with OUPUT bits 0 through 7 for each port gets combined and simplified into OUTPUT bits 1 through 48. The JMRI Bit Assignments window will automatically show the 24 OUTPUT pins on the SMINI as CTXX. Note Well:  Any of the six SMINI’s  ‘output cards’ can be configured by the assembler of that particular SMINI as either ‘current sinking’ or ‘current sourcing’, and you will have to keep track of that to successfully use the SMINI outputs. You might note in the comments for individual Outputs (‘Turnouts’) whether they are configured as  Current Sinking or Current Sourcing. Note also that SMINI Current Sinking Outputs are not compatible with the SMINI inputs, and will thus cause errors in the CMRI wraparound test available in JMRI.”
 -------------------
I assume that changes or additions to the JMRI documentation require review. The above suggested revision is intended to help others to not make the same mistakes I made in trying to set up my SMINI to work with JMRI.
-- 
RRGibbons

KenC57, 2021-11-11:
I think this is way too much about just the SMINI case. The real issue is how any node deals with its I/O
boards.

When JMRI is communicating with a node, all of the input bits are concatenated together as well as all the
output bits. The mixing of the input or output cards in a node does not change the bit numbers. Both ranges
start from 1 for that node. Example the first output (turnout) is CT1 for node zero bit 1, CT1001 for node
one, etc... Same patter with inputs (sensors) as CS1 for node zero bit 1, CS1001 for node one, etc... The
lowest board on each node would be the first bit and so on.

rrgibbons, 2021-11-11:
Ken,

Thanks for responding. What do I do? Are you suggesting I figure out a more general addition to submit  to place elsewhere in the documentation, or did I just miss an explanation  in the existing documentation that you explained to me in your message below?

Regards,

David Gibbons

jerryg2003, 2021-11-11:
David

What you say is true for SUSICs too (and perhaps for all C/MRI node types).  I use C/MRI via simulated nodes on arduinos and I just define them all as SUSICs with however many input and output lines I want (in increments of Card Size: 8, 16, or 24).   

Also, as I understand it, Lights and Turnouts share the same “space” in terms of C/MRI outputs.  So CT001 and CL001 cannot both exist simultaneously.   

If you want to take another pass at a paragraph, perhaps to go earlier in the sections so that it applies to all node types, I’m happy to update the web/help page for you, or show you how.  Always glad to have someone else interested in updating out documentation!

Jerry

KenC57, 2021-11-11:
To David and Jerry,

Yes I was meaning think of distilling the text and making the point it is general how the I/O is presented in
JMRI. Trying to think of a concise way to say this may not be ease, but the amount of text for your
explanation 'just for SMINI' seemed overkill. Figuring how to make it apply to all the board combinations is
the best for all I think.

And just trying to explain the part of turnouts vs lights, there's a tricky one too. How to say that a given
number for an output may only be for a turnout OR light is important and sometimes missed. That little point
from Jerry should be in this somewhere.

rrgibbons, 2021-11-12:
Gentlemen,

When Ken speaks of making the point general, does this point apply ONLY to all of the C/MRI hardware and hardware emulators, or are there also other hardware I/O technologies where the point is also applicable?

If we are only concerned with C/MRI, then I would suggest that the documentation section addressing “Managing Nodes” or “Configuring C/MRI Nodes” could be the place to add my contribution. The text just above the image of the node configuration manager seems appropriate to me. 

I will review the text I proposed with that location in mind, and submit that for your consideration.

Regards,

David Gibbons

rrgibbons, 2021-11-13:
> David G,
>
> I would think focusing on CMRI is the reasonable approach. It is one of the few protocols where we have this
> detailed node configuration data and variable at that. How JMRI tries to represent all those different
> possible combinations into a reasonable consistent view deserves the explanation. I would say most other
> connections don't really keep a detail of the external hardware like we do for CMRI. In a LocoNet connection
> the user manages that a given board creates some range of turnouts and sensors and JMRI doesn't know if that
> is one board or five. Due to how we talk to CMRI nodes, we need this extra data.
>
> Long term I would love to see us find a way to remove the node information from inside the connection entry.
> Currently if you change your connection, like live to simulator, it loses the node details most of the times.
> Many times I resorted to editing the profile to cut/paste the node information into it when these changes lost
> the data. But nobody who understands that part of the code enough has enough time to do that sort of work.
> Luckily most users don't make the types of changes in their connection that loses the node data. Or they
> happily just re-enter it when this does happen.
>
> -ken c

Gentlemen,

How about this revised version of the proposed text?

“When setting up C/MRI or C/MRI-compatible  I/O boards in JMRI, be aware that JMRI simplifies things, putting all input (‘sensor’) pins in a series from 1 to the maximum number of ‘IN bits’ for that board using the prefix ‘CS’. All output pins (‘turnouts’) will be handled similarly, using the prefix ‘CT’ for the ‘OUT Bits'. If you have 2 or more boards set up as C/MRI nodes, the prefix gets an additional character added after the ‘C’ to indicate the additional connected nodes. Note Well:  Some C/MRI output boards can be configured in whole or in part as either ‘current sinking’ or ‘current sourcing’, and you will have to keep track of that to successfully use the outputs. Note also that Current Sinking Outputs are not compatible with the inputs, and will thus cause errors in the CMRI wraparound test available in JMRI.”

This revised text could be plugged into the block of text below the “Configuring C/MRI Nodes” heading, aove the ‘Node configuration Manager’ image.

Regards,

David Gibbons

jerryg2003, 2021-11-15:
David -  To help this conversation, I took the liberty of creating a modified version of the page (see Pull Request #10315).  You will see that I put the last part of your suggestion (which I have taken verbatim since I'm not familiar with the issue) in the section on Supported Hardware and put a paraphrased version of the first part  at the end of the Configuring C/MRI Nodes section.  You can download this by going to the PR.  I marked this as "Work In Progress" so it will not be incorporated into the release until we are done discussing.  If you prefer different wording or placement, you can edit the HTML and resubmit, or I'll be happy to upload for you.  Hope you become an active participant in continuing to improve the JMRI documentation!

bobjacobsen, 2021-11-15:
To see Jerry's update as a web page, not counting the top/side/bottom nav info:  

https://htmlpreview.github.io/?https://github.com/JMRI/JMRI/blob/fe73c920822005cacd1831a926ce7021d39127fd/help/en/html/hardware/cmri/CMRI.shtml

jerryg2003, 2021-11-19:
Having heard no further comments, PR #10315 has been merged.

closed 2021-11-19 by jerryg2003