JMRI/JMRI #10293: Single CV Programmer labelling deficiency leads to user confusion.

dheap, 2021-11-12:
The Single CV Programmer doesn't mention the user-familiar "Program Track" or "Program On Main" terms.

A colleague over here in Oz was confused and asked me about it in the past week. Now we have the same confusion reported in a [jmrusers thread](https://groups.io/g/jmriusers/topic/86975370#196346).

The [graphic](https://www.jmri.org/help/en/package/jmri/jmrit/simpleprog/SimpleProgFrame.jpg) also needs updating. It predates the enhanced Accessory Ops Mode options now available.

danielb987, 2021-11-12:
[image]

The image shows the current Single CV Programmer dialog. What should be changed?

I seem to have two LocoNet connections in this dialog, despite only having one LocoNet connection configured in JMRI (LocoNet simulator).

bobjacobsen, 2021-11-12:
If I were coding this again, I’d move the “Program on Main” over the connection-selector, and add a similar “Programming Track” in the box above that.  If ambitious, I’d use a titled border instead of an embedded label.

danielb987, 2021-11-14:
[image]

See PR #10308

I have a couple of problems:
The text `Programming On Main` works, but the text `Programming Track` is cut. How do I get this text to be shown in its entirety?

These classes uses `java.util.Vector` and it would be good to replace that with `java.util.List`. But if I change
`java.util.Vector<GlobalProgrammerManager> v = new java.util.Vector<>();`
to
`java.util.List<GlobalProgrammerManager> v = new java.util.ArrayList<>();`
I get the error `cannot infer type arguments for JComboBox<>` on this line:
`add(progBox = new JComboBox<>(v));`

How should I solve this?

bobjacobsen, 2021-11-14:
> I get the error cannot infer type arguments for JComboBox<>

JCombox only has constructors from a vector or array.  I think the cleanest way is to use List's [toArray](https://docs.oracle.com/javase/8/docs/api/java/util/List.html#toArray-T:A-) method to create a temporary array:

```
new JComboBox<>(v.toArray(new Array< GlobalProgrammerManager >[0]);
```

bobjacobsen, 2021-11-14:
It looks like the upper programming pane is being embedded without a LayoutManager.  Take a look at how the bottom (ops mode) pane is added and check for differences.  Sometimes you have to add a nested pane (put the content pane inside another one) or have nested LayoutManagers.

bobjacobsen, 2021-11-14:
I was wrong about that.  The upper box it being embedded with a BoxLayoutManager, which pays (too much) attention to included component width.

The answers in this [StackOverFlow Q&A](https://stackoverflow.com/questions/6407897/title-truncation-in-jpanel-titledborder-java-swing) might be helpful. Note the comment on the 2nd one.

danielb987, 2021-11-15:
> The answers in this [StackOverFlow Q&A](https://stackoverflow.com/questions/6407897/title-truncation-in-jpanel-titledborder-java-swing) might be helpful. Note the comment on the 2nd one.

Thanks! I have updated the PR and the dialog now looks like this. Note that I have never used `Simple programmer` so I haven't tested the programming feature. I have only checked how it looks on the screen.

[image]

closed 2021-11-16 by bobjacobsen