JMRI/JMRI #10294: Java exception when C/MRI connection lost

jerryg2003, 2021-11-12:
JMRI 4.25.4 on Windows 10.  Disconnected USB cable to RS-485 network for my C/MRI nodes (arduinos) and got red message on main PanelPro window (good) and this in the system console (bad).  When I tried to plug the USB back in, nothing happened and I had to quit and restart JMRI.

jtermios.windows.JTermiosImpl$Fail
	at jtermios.windows.JTermiosImpl$Port.fail(JTermiosImpl.java:100)
	at jtermios.windows.JTermiosImpl.select(JTermiosImpl.java:924)
	at jtermios.JTermios.select(JTermios.java:474)
	at purejavacomm.PureJavaSerialPort$2.read(PureJavaSerialPort.java:882)
	at java.io.DataInputStream.read(Unknown Source)
	at jmri.jmrix.AbstractMRTrafficController.readByteProtected(AbstractMRTrafficController.java:957)
	at jmri.jmrix.cmri.serial.SerialTrafficController.waitForStartOfReply(SerialTrafficController.java:333)
	at jmri.jmrix.AbstractMRTrafficController.handleOneIncomingReply(AbstractMRTrafficController.java:1058)
	at jmri.jmrix.AbstractMRTrafficController.receiveLoop(AbstractMRTrafficController.java:878)
	at jmri.jmrix.AbstractMRTrafficController$2.run(AbstractMRTrafficController.java:801)
	at java.lang.Thread.run(Unknown Source)
2021-11-12 07:50:09,113 jmrix.AbstractMRTrafficController     ERROR - run: Exception: java.io.IOException in class jmri.jmrix.cmri.serial.SerialTrafficController [serial.SerialTrafficController Receive thread]
java.io.IOException
	at purejavacomm.PureJavaSerialPort$2.read(PureJavaSerialPort.java:884)
	at java.io.DataInputStream.read(Unknown Source)
	at jmri.jmrix.AbstractMRTrafficController.readByteProtected(AbstractMRTrafficController.java:957)
	at jmri.jmrix.cmri.serial.SerialTrafficController.waitForStartOfReply(SerialTrafficController.java:333)
	at jmri.jmrix.AbstractMRTrafficController.handleOneIncomingReply(AbstractMRTrafficController.java:1058)
	at jmri.jmrix.AbstractMRTrafficController.receiveLoop(AbstractMRTrafficController.java:878)
	at jmri.jmrix.AbstractMRTrafficController$2.run(AbstractMRTrafficController.java:801)
	at java.lang.Thread.run(Unknown Source)
2021-11-12 07:50:09,129 jmrix.AbstractMRTrafficController     ERROR - Exit from rcv loop in class jmri.jmrix.cmri.serial.SerialTrafficController [serial.SerialTrafficController Receive thread]

[Note: MQTT disconnect in Issue #10047 throws different exception.  Fix to that being worked on PR #10291.  Do other connection types have similar "non-graceful" exits on disconnect?]

bobjacobsen, 2021-11-12:
> Do other connection types have similar "non-graceful" exits on disconnect?

Probably most of them.  And probably most of them, including this one, would be hard to fix completely; there are lots of different states to deal with, and you have to do it way down in the comm/USB stack.  I think the most appropriate action on this is "don't do that".

bobjacobsen, 2021-11-12:
Just saw the MQTT note in PR#10291

bobjacobsen, 2021-11-12:
Just saw the MQTT note in PR#01921.

Pulling a USB connection for any serial protocol is not analogous to MQTT losing a connection.  Pulling a USB hardware unit is a _hardware_ level fault.  The equivalent for MQTT would be yanking the ethernet card (don't try this at home).  The equivalent here would be disconnecting or powering down a C/MRI node, which JMRI survives just fine.

danielb987, 2021-11-12:
If this PR is about pulling the USB connection while JMRI is running, I suggest this Issue is closed. There is no reason to why JMRI should be able to handle pulling and reconnecting the USB connection while running.

closed 2021-11-14 by jerryg2003