JMRI/JMRI #10301: Request to add TypeScript support to build scripts

geowar1, 2021-11-13:
I've often thought about refactoring /web/js/panel.js to parallel the LayoutEditor classes. But if I'm going to go that far I'd prefer to convert it to TypeScript at the same time (to pickup the benefits of TypeScript). I tried just changing the extension to .ts but JMRI's build system didn't know (how?) to translate (compile?) that to panel.js.
Not knowing the build system I wouldn't know where to start to fix this… does anyone else?
Little help?

bobjacobsen, 2021-11-14:
There are two possible approaches to this, depending on what we want to require of all developers:

1) Have ant and maven (hence IDE) targets that do the TypeScript -> .js every time. Everybody would need to be able to do that, hence have tools, etc

2) As above, except store the compiler-output .js into Git so that only the people working on TypeScript need to have the tools.  (This is similar to how we handle the launchers)

I think we can add an ant target pretty easily to do either of those pretty easily.  I'm not sure what it would take to configure maven and hence IDEs.  

If (2) sounds OK, I'll start on the target.

danielb987, 2021-11-14:
For me, (2) is preferred. When JMRI went from ant to Maven, it caused problems with NetBeans. If every developer is required to have the tools to do TypeScript, that might cause problems that we might have difficult to solve.

bobjacobsen, 2021-11-14:
@danielb987 Good point. 

@geowar1:  If you're OK with (2), I recommend creating a `web/ts/` parallel to `web/js/`.  Then any .ts file in `web/ts/` will get compiled to the parallel directory in `web/js` by a separate Ant step.  Would probably want to have a conventional top-comment in the .ts files that makes the compiler-output .js files distinctive.  (Adding one automatically gets complicated). 

Let me know if I should proceed on this, please.

danielb987, 2021-11-14:
Sounds good to me.

geowar1, 2021-11-15:
On Nov 14, 2021, at 4:28 AM, Bob Jacobsen ***@***.***> wrote:
> @geowar1: If you're OK with (2), I recommend creating a web/ts/ parallel to web/js/. Then any .ts file in web/ts/ will get compiled to the parallel directory in web/js by a separate Ant step. Would [...]
I was about to suggest the parallel web/ts directory. The transcribe tool would then create a new js file in the web/js directory for each ts file in the web/ts directory.

I don't know enough about how TS handles modules to know how to implement them; just that the transcribe step could take several .ts files to create a single .js file. Perhaps the TS (sub-)modules would(/should?) be in subdirectories and the transcribe would only create parallel JS files for the TS files in the web/ts root directory?

Sent from GeoWar’s iPad Pro

bobjacobsen, 2021-11-15:
Take a look at my bobjacobsen/add-TypeScript branch.  It has some initial support with `ant typescript` for compiling from web/ts -> web/js.  I put a HelloWorld example there with a proposed top comment.

There's a `tsconfig.json` file in web/ts that has a _lot_ of build options that I only dimly understand.  It looks like there are several options for how the .ts files are mapped to .js.  I looks like it's going to take a bit of experimentation to figure out the best (available) approach.

Let me know if this works for you, please, and I'll PR it.

danielb987, 2021-11-15:
I have never worked with TypeScript, so this is only guesses.

> There's a `tsconfig.json` file in web/ts that has a _lot_ of build options that I only dimly understand.

Don't know if this could be useful:
```
// "paths": {},       /* Specify a set of entries that re-map imports to additional lookup locations. */
// "typeRoots": [],   /* Specify multiple folders that act like `./node_modules/@types`. */
```
I don't know if this should be enabled?
```
/* JavaScript Support */
// "allowJs": true,              /* Allow JavaScript files to be a part of your program. Use the `checkJS` option to get errors from these files. */
// "checkJs": true,              /* Enable error reporting in type-checked JavaScript files. */
// "maxNodeModuleJsDepth": 1,    /* Specify the maximum folder depth used for checking JavaScript files from `node_modules`. Only applicable with `allowJs`. 
```

danielb987, 2021-11-19:
TypeScript is now supported so I'm closing this issue.

closed 2021-11-19 by danielb987