JMRI/JMRI #10318: XML for new LDH decoders

ohojvat, 2021-11-15:
There are two files included

Working with winXP I  can´t see if .rar file is well attached, but the name is shown when I  place the cursor on the regarding zone. 

One for spa , other for portugues Labeled ...spa and ...prtg

This was traslated to portuguese in Brasil so we can't do only one file

Anyway now the total number of models are only 6

Best regards, Osvaldo Hojvat

danielb987, 2021-11-15:
@ohojvat 
Thanks. But the rar file is not attached.

bobjacobsen, 2021-11-15:
.rar isn't a supported format. I've requested a direct email of the files and will attach them.

ohojvat, 2021-11-15:
Here are attached the XML files

Thank you

ohojvat, 2021-11-15:
Mail sent to Bob  with 2 XML files
and I try to attach here .ZIP file

ohojvat, 2021-11-16:
[decodersV60.zip](https://github.com/JMRI/JMRI/files/7547782/decodersV60.zip)

ohojvat, 2021-11-16:
now with W10 the .zip file seems to be attached 

Please confirm

bobjacobsen, 2021-11-16:
.zip file present.

Are these two language versions of the exact same decoder definition?  If so, I can merge them into one using our language support so that the language gets automatically selected.  Could you confirm what your version of JMRI shows for the language code?

[image]

[image]

ohojvat, 2021-11-16:
Thak you for your help.

Sebastian Burone, in Brazil, is using english Java for the portuguese texts version

When seting portuguese, portuguese_Brazil or portuguese_Portugal he had no change in JMRI texts
So he also use english in preferences.

At my side, I have spanish java and also have spanish in preferences. Most JMRI changes

There are references in our texts about the name of sheets so if lang. may be selected, some references will be wrong.

As we are a very small team: 1 man in Brasil and 2 people in Argentina, we have not more available time for this XML matter.

We think the best at this moment will be the two xml be added.
This decoder model in hardware the same of the V55 one  in actual JMRI
 
But the software has many improvements.
Using the same small uP  (pic16f1825) it has sintetized sound for 6 diferents locomotive types, 12 horns abd 4 whistles, bell and some other sounds, all selected by CVs

Next year will be released a new model with recorded sounds, with a little biger uP.
I must return to this programming work now

We will work in this case on the xml files with the two language versions from the start.

And MrBurone needs to needs to return working with production an sell

Regards, Osvaldo Hojvat

bobjacobsen, 2021-11-16:
Thank you!  Appreciate your work on these.  We’ll get them merged into JMRI.

ohojvat, 2021-11-17:
Thank you  

Mr. Sebastian had made several youtube videos about using JMRI,  in portuguese
So when offering our new decoder version, some people ask him about the decoder definition for JMRI.  

The mentioned videos are in the following links; may be someone more.

https://youtu.be/gGuH7DOAGn4

https://youtu.be/-iRQBAmOXxA

https://www.youtube.com/watch?v=Vj39PzCEkrk

https://www.youtube.com/watch?v=VkxgZfkLBkY

https://youtu.be/6OEeV7gTlNE

https://youtu.be/oW3jK2YSOTs
 Regards, Osvaldo

closed 2021-11-19 by danielb987

ohojvat, 2022-01-10:
Dear Mr Jacobsen
 
Thank you for your help in other matters.

We (LDH) are makeing from many years a Command Station  LDH35  and a (new product) PC interfase ITX-PC 
We state in JMRI preferences LENZ   and  Lenz LI-101  thats are very similar to our parts

For a before issue 10318, I know that it will be a lot of work for showing our names in preferences and adjust all for our small diferences.

I am at the moment requesting a very small job (I guess) about System Information panel 

We are using from january 2022 the internal number 7 so the reply for  Request system information packet: 21 21 00
is packet: 63 29 39 07 0C

and it is shown: 
  Estación de Comandos<unknown>  
  Software version 3.9
   ...

Can be modified this  <unknown>  label by  our command station  name     LDH-35        ?

The internal number 7 may be changed  if it is necessary

Best Regards, Osvaldo Hojvat

ohojvat, 2022-01-12:
Could you please try one of the installers at:

https://builds.jmri.org/jenkins/job/development/job/packages/876/

I think I’ve made your change properly, but I have no way to test it.

Bob

> 
> Dear Mr Jacobsen
> 
> Thank you for your help in other matters.
> 
> We (LDH) are makeing from many years a Command Station  LDH35  and a (new product) PC interfase ITX-PC
> 
> We state in JMRI preferences LENZ   and  Lenz LI-101  thats are very similar to our parts
> 
> For a before issue 10318, I know that it will be a lot of work for showing our names in preferences and adjust all for our small diferences.
> 
> I am at the moment requesting a very small job (I guess) about System Information panel 
> 
> We are using from january 2022 the internal number 7 so the reply for  Request system information packet: 21 21 00
> is packet: 63 29 39 07 0C
> 
> and it is shown: 
>   Estación de Comandos<unknown>  
>   Software version 3.9
>    ...
> 
> Can be modified this  <unknown>  label by  our command station  name     LDH-35        ?
> 
> The internal number 7 may be changed  if it is necessary
> 
> Best Regards, Osvaldo Hojvat
> 
> 
> 
> 
> 
> 
> Thank you! Appreciate your work on these. We’ll get them merged into JMRI.

bobjacobsen, 2022-01-13:
Sorry.  Hopefully fixed.  Could you try again please?

Bob

> 
> 
> Could you please try one of the installers at:
> 
> https://builds.jmri.org/jenkins/job/development/job/packages/876/
> 
> I think I’ve made your change properly, but I have no way to test it.
> 
> Bob
> 
> > 
> > Dear Mr Jacobsen
> > 
> > Thank you for your help in other matters.
> > 
> > We (LDH) are makeing from many years a Command Station LDH35 and a (new product) PC interfase ITX-PC
> > 
> > We state in JMRI preferences LENZ and Lenz LI-101 thats are very similar to our parts
> > 
> > For a before issue 10318, I know that it will be a lot of work for showing our names in preferences and adjust all for our small diferences.
> > 
> > I am at the moment requesting a very small job (I guess) about System Information panel 
> > 
> > We are using from january 2022 the internal number 7 so the reply for Request system information packet: 21 21 00
> > is packet: 63 29 39 07 0C
> > 
> > and it is shown: 
> > Estación de Comandos<unknown> 
> > Software version 3.9
> > ...
> > 
> > Can be modified this <unknown> label by our command station name LDH-35 ?
> > 
> > The internal number 7 may be changed if it is necessary
> > 
> > Best Regards, Osvaldo Hojvat
> > 
> > 
> > 
> > 
> > 
> > 
> > Thank you! Appreciate your work on these. We’ll get them merged into JMRI.
> > 
> 
> Bob Jacobsen

—
Bob Jacobsen
***@***.***

ohojvat, 2022-01-13:
I guess it will work with w10

Will try now

 Which one at first ?
Osvaldo 

 
 

—
Bob Jacobsen
***@***.***

ohojvat, 2022-01-13:
I will try the second one with w10 notebook

waitin the download

 
 

—
Bob Jacobsen
***@***.***

ohojvat, 2022-01-13:
I tryed to download JMRI.4.25.10ish+jenkins+20220112T1709Z+Race3bda3d5.exe four times but it fails at about  19 minutes of 50 minutes expected download time

I will try again later

Osvaldo 

 
 

—
Bob Jacobsen
***@***.***

ohojvat, 2022-02-07:
About XML for new LDH decoders, I have tested JMRI version 4.26 and found the XML files working OK, thank you

-------------------------------------------------------

About the LDH name added ,  I try downloading again , after your fixed ,but it begins downloading but about 98% it stops. It tryed several times with a diferent PC and have the same problem

But I found the regarding file you have modified and it seems to be OK  (line 97 -in the attached pdf copy)

Please give me a link for the file I need to install for the test

I was on a trip and now I returned with this matter

Best Regards, Osvaldo Hojvat

 
 

—
Bob Jacobsen
***@***.***

ohojvat, 2022-02-07:
----- Mensaje reenviado ----- De: Osvaldo Hojvat ***@***.***>Para: JMRI/JMRI ***@***.***>; Bob Jacobsen ***@***.***>Enviado: domingo, 6 de febrero de 2022 22:53:53 ARTAsunto: Re: [JMRI/JMRI] XML for new LDH decoders (Issue #10318) and LDH name in system information
  About XML for new LDH decoders, I have tested JMRI version 4.26 and found the XML files working OK, thank you

-------------------------------------------------------

About the LDH name added ,  I try downloading again , after your fixed ,but it begins downloading but about 98% it stops. It tryed several times with a diferent PC and have the same problem

But I found the regarding file you have modified and it seems to be OK  (line 97 -in the attached pdf copy)

Please give me a link for the file I need to install for the test

I was on a trip and now I returned with this matter

Best Regards, Osvaldo Hojvat

 
 

—
Bob Jacobsen
***@***.***

ohojvat, 2022-03-22:
Dear Mr. Jacobsen

I can´t download the mentioned installers:
.......................

I think this small changes will work fine 

Please include it in future JMRI version or please let me know if it is done.

If it is not, and if you prefer, please send me a test version , but in the form of a normal version to be download.

 
 

—
Bob Jacobsen
***@***.***

danielb987, 2022-03-22:
> Dear Mr. Jacobsen I can´t download the mentioned installers:

Try this:
https://builds.jmri.org/jenkins/job/development/job/packages/924/

bobjacobsen, 2022-03-22:
The changes are in the most recent JMRI test release.

https://www.jmri.org/releasenotes/jmri4.99.4.shtml

ohojvat, 2022-03-24:
ts is all rigth
Thank you Bob and Daniel
Best Regards Osvaldo

 
 

https://www.jmri.org/releasenotes/jmri4.99.4.shtml