JMRI/JMRI #10371: Java exception during startup

jerryg2003, 2021-11-23:
Using Windows 10.  Just started happening after I did some changes to a layout where I deleted some memory variables.  Closed and reopened PanelPro and got this (happened in 4.25.5 so I installed 4.25.6 and it still happened) before I loaded any panels or scripts:

JMRI 4.25.6 on Windows 11:

2021-11-23 17:50:53,767 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2021-11-23 17:50:53,767 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\jerry\JMRI\log\messages.log [main]
2021-11-23 17:50:53,767 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\jerry\JMRI\log\session.log [main]
2021-11-23 17:50:53,783 apps.Apps                             INFO  - PanelPro version 4.25.6+R56a369f047 starts under Java 11.0.13 on Windows 10 amd64 v10.0 at Tue Nov 23 17:50:53 EST 2021 [main]
2021-11-23 17:50:56,089 apps.Apps                             INFO  - Starting with profile SSimCSimMQTT.3db5210d [main]
2021-11-23 17:50:56,305 node.NodeIdentity                     INFO  - Using 41de1639-e2be-42ad-a792-ecc1326eea78 as the JMRI storage identity for profile id 3db5210d [AWT-EventQueue-0]
2021-11-23 17:50:56,367 managers.JmriUserPreferencesManager   ERROR - setClassDescription(jmri.jmrit.beantable.MemoryTableDataModel) failed in newInstance [AWT-EventQueue-0]
java.lang.NoSuchMethodException: jmri.jmrit.beantable.MemoryTableDataModel.<init>()
	at java.base/java.lang.Class.getConstructor0(Class.java:3349)
	at java.base/java.lang.Class.getDeclaredConstructor(Class.java:2553)
	at jmri.managers.JmriUserPreferencesManager.setClassDescription(JmriUserPreferencesManager.java:613)
	at jmri.managers.JmriUserPreferencesManager.setMultipleChoiceOption(JmriUserPreferencesManager.java:819)
	at jmri.managers.JmriUserPreferencesManager.lambda$12(JmriUserPreferencesManager.java:960)
	at java.base/java.util.Iterator.forEachRemaining(Iterator.java:133)
	at java.base/java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1801)
	at java.base/java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:658)
	at jmri.managers.JmriUserPreferencesManager.lambda$11(JmriUserPreferencesManager.java:953)
	at java.base/java.util.Iterator.forEachRemaining(Iterator.java:133)
	at java.base/java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1801)
	at java.base/java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:658)
	at jmri.managers.JmriUserPreferencesManager.lambda$10(JmriUserPreferencesManager.java:952)
	at java.base/java.util.Iterator.forEachRemaining(Iterator.java:133)
	at java.base/java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1801)
	at java.base/java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:658)
	at jmri.managers.JmriUserPreferencesManager.readPreferencesState(JmriUserPreferencesManager.java:949)
	at jmri.managers.JmriUserPreferencesManager.readUserPreferences(JmriUserPreferencesManager.java:865)
	at jmri.managers.JmriUserPreferencesManager.initialize(JmriUserPreferencesManager.java:1158)
	at jmri.InstanceManager.getInstance(InstanceManager.java:326)
	at jmri.InstanceManager.getNullableDefault(InstanceManager.java:233)
	at jmri.InstanceManager.getDefault(InstanceManager.java:204)
	at apps.Apps.lambda$0(Apps.java:186)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:303)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
2021-11-23 17:50:56,436 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "SPROG Sim" [main]
2021-11-23 17:50:56,490 simulator.SimulatorAdapter            INFO  - SPROG Simulator Started [SPROG Simulator]
2021-11-23 17:50:56,937 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "C/MRI" [main]
2021-11-23 17:50:58,278 script.JmriScriptEngineManager        INFO  - ECMAScript ECMA - 262 Edition 5.1 is provided by Oracle Nashorn 11.0.13 [AWT-EventQueue-0]
2021-11-23 17:50:58,278 script.JmriScriptEngineManager        INFO  - python 2.7 is provided by jython 2.7.2 [AWT-EventQueue-0]
2021-11-23 17:50:58,541 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2021-11-23 17:50:58,541 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\jerry\Dropbox (Personal)\JMRI\My_JMRI_Railroad\ [main]
...normal lines after this...

danielb987, 2021-11-23:
The exception is well known and doesn't cause any major problems. We are aware of the problem and that the code needs to be rewritten. I have thought about doing it but haven't yet had time to do it.

But this Issue is useful since it shows that the problem is more widespread than I first thought. So I would like this Issue to stay open.

The cause of the exception is that the class `jmri.jmrit.beantable.MemoryTableDataModel` doesn't have a constructor that doesn't take any arguments.

dsand47, 2021-11-23:
As Daniel mentioned, the user-interface.xml file in profile/41de1639-e2be-42ad-a792-ecc1326eea78 has a reference that cannot be resolved.

The easy solution is to delete file, but you lose window locations and other saved settings.  The alternative is edit the xml and remove the bad entry (while JMRI is not running of course).

jerryg2003, 2021-11-27:
Thanks.  Did some more sleuthing in case this is useful:  the Java exception happens iff I attempt to delete a Memory AND check the "remember this setting for next time" box on the delete yes/no dialog popup window.  This causes lines to appear in userinferface.xml under the classPreferences section:

```
        <preferences xmlns="" class="jmri.jmrit.beantable.MemoryTableDataModel">
            <multipleChoice>
                <option item="deleteInUse" value="2"/>
            </multipleChoice>
        </preferences>

```
If I delete those lines, the Java exception does not appear on the next invocation of JMRI.  Deleting (or not) the lines about Memory tables in the "tabeDetails" section does not have any effect regarding the Java exception.

bobjacobsen, 2021-12-01:
I.m working on this.

Bob

> Thanks. Did some more sleuthing in case this is useful: the Java exception happens iff I attempt to delete a Memory AND check the "remember this setting for next time" box on the delete yes/no [...]

—
Bob Jacobsen
***@***.***

danielb987, 2022-01-15:
This particular case has been fixed, but the underlying problem should be fixed.

closed 2022-03-17 by stale[bot]