JMRI/JMRI #10412: No support for DCC++ EX "JOIN" track power state

JDeeth, 2021-12-01:
Hi,

DCC++ EX allows track power to be separately controlled for the main and programming track, and also has a "JOIN" function where DCC signals are also sent to the programming track. This permits the programming track to be physically connected to the rest of the layout, and trains to be driven on and off the section under their own steam.

I'd like to have a go at adding support for this feature to JMRI and see how I get on. (Though realistically it'd be much easier to use a DPDT switch or equivalent, or send the command to the base station via another Arduino!)

danielb987, 2021-12-01:
@JDeeth 
Welcome to the JMRI project!

@mstevetodd 
Can you look at this issue? It seems like an interesting issue to solve and it would be good to give guidance to @JDeeth to implement it. I'm happy to help, but you have much more knowledge on DCC++ EX.

JDeeth, 2021-12-01:
Cheers :)
DCC++ has `<1>` and `<0>` commands to turn on all track power at once. DCC++ EX supports this too but allows optional parameters.

My experience is onl y with DCC++ - do other systems support separate control of main and programming track power?

danielb987, 2021-12-01:
> My experience is onl�y with DCC++ - do other systems support separate control of main and programming track power?

I don't know, but I don't think so. But JMRI supports that a connection have a connection specific menu and DCC++ have it's own menu. On that menu, you could add a DCC++ EX specific power control frame, similar to the `Tools -> Power Control` frame. That DCC++ EX specific power control frame could have buttons for Power On / Power Off for both main and programming track, and a button for Join.

An example dialog on how it could be done:

[image]

mstevetodd, 2021-12-02:
I responded to what I think is the same question on DCC-EX's Discord group.

@JDeeth I'm pretty sure the programming track power state is a DCC-EX-only concept, so the "base" JMRI doesn't have any support for it. As Daniel said, you could code something for DCC-EX-only, or (maybe?) enhance JMRI to have this concept.

JDeeth, 2021-12-02:
Yup, that's me in the Discord - someone has reminded me you can send `<1 JOIN>` to the base station via "Send DCC++ Command" already in the DCC++ module.

closed 2022-03-20 by stale[bot]