JMRI/JMRI #10428: Mostly minor LogixNG issues/Java exception

jerryg2003, 2021-12-03:
Doing some more experimentation with this excellent addition to JMRI.  Found the following issues in 4.25.7 on Windows 10 with Azul Java 11.0.13:

1. "Browse" does not list the "Local variable" statement that appears in my actual LogixNG.  (I have it right after the initial "A Many" statement).
2. If I use the "copy" pulldown item on a fairly complex conditional where I have previously edited a number of things (including removing various sockets) and then "paste" it into a new (empty) conditional,  the removed sockets show up again (I have the example if you need it).
3. Continuing to edit that complex conditional, I started to "cut" things and then "remove the socket," and after about 4 edits got the Java exception shown below.
4. A LogixNG that begins with an "If Then Else. Execute on Change" (copied below), where the "if" checks if a sensor is active, is triggered on JMRI startup.  The sensor still shows "Unknown" so to my mind has NOT "changed."  Why did the LogixNG get triggered (the LogixNG correctly sees the sensor as still being Unknown)?
5. Same LogixNG, if I unclick Enable, it is not triggered at startup, but it is triggered as soon as I click Enabled.  Why?
6. Same LogixNG, when I click the sensor it goes from Unknown to Inactive, but the logic is NOT triggered (I expected that it would be triggered because the sensor changed).  When I click again to Active, it is triggered (as expected).  When I click again to Inactive, it is triggered again (as expected).  Why isn't it triggered on going from Unknown to Inactive?

Example LogixNG for items 4, 5, and 6 (needs sensor ISABC and memory IMABC):
LogixNG: Test
    ConditionalNG: IQC:AUTO:0003
        ! A
            If Then Else. Execute on change
                ? If
                    Sensor ISABC is Active
                ! Then
                    Set memory IMABC to "Sensor CHANGE - Active"
                ! Else
                    If Then Else. Always execute
                        ? If
                            Sensor ISABC is Inactive
                        ! Then
                            Set memory IMABC to "Sensor CHANGE - Inactive"
                        ! Else
                            Set memory IMABC to "Sensor change - not Active or Inactive"

Java exception for item 3 above:
2021-12-02 22:11:43,347 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.IllegalArgumentException: Invalid index
	at jmri.jmrit.logixng.FemaleSocket.isSocketOperationAllowed(FemaleSocket.java:126)
	at jmri.jmrit.logixng.tools.swing.TreeEditor$PopupMenu.showPopup(TreeEditor.java:1526)
	at jmri.jmrit.logixng.tools.swing.TreeEditor$PopupMenu.access$1(TreeEditor.java:1476)
	at jmri.jmrit.logixng.tools.swing.TreeEditor$PopupMenu$1.openPopupMenu(TreeEditor.java:1467)
	at jmri.jmrit.logixng.tools.swing.TreeEditor$PopupMenu$1.mouseReleased(TreeEditor.java:1450)
	at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:298)
	at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:297)
	at java.desktop/java.awt.Component.processMouseEvent(Component.java:6635)
	at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342)
	at java.desktop/java.awt.Component.processEvent(Component.java:6400)
	at java.desktop/java.awt.Container.processEvent(Container.java:2263)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5011)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

danielb987, 2021-12-03:
Thanks for your comments!

> 1. "Browse" does not list the "Local variable" statement that appears in my actual LogixNG.  (I have it right after the initial "A Many" statement).

Open the Browse window. At the bottom, you have four checkboxes.
- Print line numbers: This shows the line number for each row in the LogixNG. This is very useful when someone asks for help, since we can then point out the exact line number where the problem lies.
- Print error handling: LogixNG has robust error handling. The combo box selection provides the default setting. Individual expressions and actions can override the default setting. This option shows the individual error handling for each expression and action.
- Print not connected sockets: Print also the sockets that aren't connected.
- Print local variables: Print the local variables that each expression and action declares.

> 2\. If I use the "copy" pulldown item on a fairly complex conditional where I have previously edited a number of things (including removing various sockets) and then "paste" it into a new (empty) conditional,  the removed sockets show up again (I have the example if you need it).

Yes, your example would be very useful.

> 4\. A LogixNG that begins with an "If Then Else. Execute on Change" (copied below), where the "if" checks if a sensor is active, is triggered on JMRI startup.  The sensor still shows "Unknown" so to [...]

IfThenElse doesn't know if the sensor has been changed. It only knows if the expression has been change since the last evaluation. And when JMRI starts, no evaluation has yet occurred so IfThenElse always do the evaluation when JMRI starts.

> 5\. Same LogixNG, if I unclick Enable, it is not triggered at startup, but it is triggered as soon as I click Enabled.  Why?

Why not? I have always thought that this was the desired behaviour.

> 6\. Same LogixNG, when I click the sensor it goes from Unknown to Inactive, but the logic is NOT triggered (I expected that it would be triggered because the sensor changed).  When I click again to [...]

As I said before, the IfThenElse action doesn't know if the sensor is changed. It knows if the expression has changed. If you want it to always evaluate, change it to `Always execute`.

jerryg2003, 2021-12-05:
@danielb987 
1. Of course!  I just have to remember to look for options at the bottom of the window...
2. Sent you an email with a panel xml.
3. You have the exception in the last post.
4. , 5., - I now understand how it is supposed to work.  If it is OK with you, I will write some explanation to add to the LogixNG documentation (and to the Logix documentation!).
6. I still seem to have this issue.  In the file I sent you, enable "Test-Startup".  Evaluates OK.  But when I click the sensor from Unknown to Inactive, nothing happens.  If I further click to Active, it works; and then back to Inactive it works.  So is there something either wrong in my logic or in the way LogixNG evaluates a statement with Unknown that then goes to Inactive? 

Thanks.

danielb987, 2021-12-07:
> 6\. Same LogixNG, when I click the sensor it goes from Unknown to Inactive, but the logic is NOT triggered (I expected that it would be triggered because the sensor changed).  When I click again to [...]

You could achieve this by changing the IfThenElse to `Always execute`. If it is `Execute on change`, it will only execute any of its child actions if the expression changes from True to False, or from False to True. But with `Always execute`, it will always execute one of its child actions, even when the expression stays False or stays True.

danielb987, 2021-12-07:
> 2\. Sent you an email with a panel xml.

Thanks. This is by design and difficult to change. Is it important for you?

> 3\. You have the exception in the last post.

I haven't been able to get it to happen on my computer. Do you get it often or only one time?

jerryg2003, 2021-12-08:
2. Sent you an email with a panel xml.

Thanks. This is by design and difficult to change. Is it important for you?
==> Not a big deal.  

3. You have the exception in the last post.

I haven't been able to get it to happen on my computer. Do you get it often or only one time?

==> I was able to generate it several times but by the time I did, I lost track of all the edits!  I'll play with it a little more (in a few days) and see if I can come up with a repeatable sequence.

jerryg2003, 2021-12-18:
@danielb987 The following sequence works repeatedly to throw a Java exception:

1. Load the example panel file I sent you.
2. Open LogixNG #0006 to Edit.
3. Right click on the A and "Copy" the whole conditional.  Then close out of this LogixNG.
4. Click "Add" to create a new LogixNG (#0007).  Click New Conditional to create a new conditional.
5. Right click on the A and "Paste".  Should show up as  +!A Many
6. Click on the A to expand, click on the A3 to expand, click on the Then Many to expand.
7. Right click on the A3 (under Then Many) and "Cut"
8. Right click on the A3 and "Remove socket"  (the socket stays there - a bug?)
9. Right click on socket A3 and throws the Java exception.

Seems a bit complicated, but potentially is the way people will use cut/paste to form new LogixNGs so getting a Java exception may indicate some other issue.

Jerry

danielb987, 2021-12-18:
@jerryg2003 
> 1. Load the example panel file I sent you.

Can you send it to me again? I'm not sure which one you refer to. Thanks!

danielb987, 2021-12-18:
@jerryg2003 
> @danielb987 The following sequence works repeatedly to throw a Java exception:

Bug is confirmed. Thank you for the very detailed bug report!

danielb987, 2021-12-18:
@jerryg2003 
The last bug should be fixed by PR #10495.

danielb987, 2022-01-18:
@jerryg2003 
Is the items in this issue fixed? Or is there more things I haven't solved yet?

jerryg2003, 2022-01-19:
Yes!  Thank you.

closed 2022-01-19 by jerryg2003