JMRI/JMRI #10439: Release Plan

bobjacobsen, 2021-12-04:
This is the current plan toward 4.26 and beyond. Dates might be off by a day or so, pending events:

- [ ] Friday Dec 10 - test release [4.25.9](https://github.com/JMRI/JMRI/milestone/114) with _only_ items that have been discussed and approved on the jmri-developers list.  Labels on PRs will be used to keep track of that. 

   We try to keep all risk out of this one, as it’ll go directly to 4.26.  The release-4.26 branch is created immediately from 4.25.9.

   If problems that must be fixed immediately are found in 4.25.9, another test release will (perhaps) be needed and the following will be delayed by a week or so.

- [ ] Saturday Dec 11 - we go back to merging all PRs to master as usual

- [ ] Friday December 17th - nominal date for releasing [4.26](https://github.com/JMRI/JMRI/milestone/116)

- [ ] Saturday December 18th - build and release [JMRI 4.99.1](https://github.com/JMRI/JMRI/milestone/115) with Java 11 changes plus current head of master branch - first to create “plus” development builds

- [ ] Friday January 14th - the normal cycle restarts with JMRI [4.99.2](https://github.com/JMRI/JMRI/milestone/118) from master+j11 branch

- [ ] As needed - JMRI 4.26.1 with necessary fixes - prefer to do this only 3 or 4 times between January and June

- [ ] Late May - start the lockdown for a mid-June JMRI 5.0; no more 4.26.* versions

bobjacobsen, 2021-12-05:
JMRI 4.25.9 will be built on Friday Dec 10 from all PRs with label "Add to Next Test Release".  Please don't merge those PRs (or any others) to master until then.  

Once 4.25.9 is out, we'll go back to our usual process, merging the rest of the PRs that are ready to go.  

Please label as "after next production release" anything that should _not_ go into 4.25.9. That includes e.g. PRs that are too big/complex to be certain of low risk, etc.

danielb987, 2021-12-07:
What is recommended for new development? Can I base new PRs on j11master and merge it into j11master after 24 hours if approved? Or is it better to base the PR on master and wait for 4.99.1 or 4.99.2? Each works for me.

Do we want to merge existing open PRs to be merged into 4.99.1 or wait for 4.99.2?

bobjacobsen, 2021-12-07:
If you’re comfortable basing your work on `j11master`, please do.    

But it’s certainly OK if people base on `master`; that’ll be merged periodically and before releases.

In particular, if you are working on a what will become a _fix_ to 4.6, please base on master.  That way we can include it in a possible, not-yet-scheduled 4.6.1 

I think we should continue merging non-WIP PRs the same way, after passing CI and a 24-hour hold. It’s fine to merge those as we go along, no reason to wait for after 4.99.1

Bob

> What is recommended for new development? Can I base new PRs on j11master and merge it into j11master after 24 hours if approved? Or is it better to base the PR on master and wait for 4.99.1 or [...]

—
Bob Jacobsen
***@***.***

danielb987, 2021-12-17:
@bobjacobsen 
> * Saturday December 18th - build and release [JMRI 4.99.1](https://github.com/JMRI/JMRI/milestone/115) with Java 11 changes plus current head of master branch - first to create “plus” development builds

Is it time to merge j11master into master now?

For 4.99.1, there is currently only one PR ready to be approved and merged,  #10484, but it only affects tests so it doesn't matter much if it's included or not.

#10486 is approved by me, but it has a change that I'm concerned about and I haven't got a reply yet from @mstevetodd . If the change is intended, I'm happy to get the PR merged.
https://github.com/JMRI/JMRI/pull/10486#discussion_r771619274

Is milestone 4.26 intended to stay open?

bobjacobsen, 2021-12-18:
> Is it time to merge j11master into master now?

Yes.  I'll create a PR for that shortly. I probably won't wait the usual 24 hours, though. Normal procedure is to include approved PRs even if they haven't reached 24 hours.

> Is milestone 4.26 intended to stay open?

There's some clean up I still have to do before I close it.  Hopefully later today!

closed 2022-02-20 by bobjacobsen