JMRI/JMRI #10449: Norwegian signals for aspect signalling

SeaU2, 2021-12-08:
Hi, I have created xml-files and icons for Norwegian signals and would like them in new releases of JMRI.

https://en.wikipedia.org/wiki/Norwegian_railway_signaling

There are two main heads in Norway: one head with green-red-green. Two green means normal speed, no diverging. One green means diverging or difficult route.
The other main head has only red-green and is mostly used on station exit with diverging route.
A distant head might be placed on same mast as a main signal.

I can not get the 2-lamp signal to work on a diverging station exit. If I put in a 3-lamp signal instead, it works ok. For me it seems to be a bug in JMRI. Or perhaps I have made an error in "appearance-Main2.xml". Please look through.

There was many things that I had problems with understanding, and I might have missed something completely.

files for xml/signals:
[Norway2001.zip](https://github.com/JMRI/JMRI/files/7675452/Norway2001.zip)
files for preferences/resources/icons:
[NorwegianSignals.zip](https://github.com/JMRI/JMRI/files/7675453/NorwegianSignals.zip)

to be entered into xml/signals/index.shtml:

<dt><a href="Norway2001/index.shtml</a>
    <dd>Norwegian rules from 2001, found in Wiki. Main signals

bobjacobsen, 2021-12-08:
See PR #10450

bobjacobsen, 2021-12-08:
Could you say a few more words about what's going wrong with the "2-lamp signal to work on a diverging station exit"?  What do you expect to see happen, and what does happen?  Thanks.

SeaU2, 2021-12-08:
[image: image.png]
The lower light for the right turnout should be showing green.
On the left turnout all are 3-lamp.
I am using loconet simulator only.

bobjacobsen, 2021-12-08:
Sorry, the image doesn't come through on an email reply.  Could you add it here please:  https://github.com/JMRI/JMRI/issues/10449

SeaU2, 2021-12-08:
Done :-)

bobjacobsen, 2021-12-08:
Thanks!  That's a puzzle, as the left and right ends seem to be behaving differently.  Could you attach the panel file here so I can try to recreate it?  And thanks for the signal system definition.

SeaU2, 2021-12-08:
Just a test file wherethe diverging route on the right turnout allways
shows red

bobjacobsen, 2021-12-08:
@dsand47 has found some missing image files.  See his comment at https://github.com/JMRI/JMRI/pull/10450#pullrequestreview-826756538  Do you have those?

SeaU2, 2021-12-08:
> See his comment at #10450 (review)

Sorry, should have checked better. Forgot about the "Large" too.

bobjacobsen, 2021-12-09:
Thanks.  I've merged those.  We're still missing Large/2and2-20B.gif though.

SeaU2, 2021-12-09:
[image: 2and2-20B.gif]

bobjacobsen, 2021-12-09:
Sorry, the file didn't come through.  Could you append? https://github.com/JMRI/JMRI/issues/10449

bobjacobsen, 2021-12-10:
@SeaU2 Thank you!  This will be in JMRI 4.25.9 in a day or two. I haven't had time to check out the logic issue, but haven't forgotten.

dsand47, 2021-12-10:
I have looked at the XML data file.  After changing the occurrences of Norway2001 to Norway-2001, the xml data file loads ok, although there are warnings and errors.

Three of the masts (LeftDist, EasSt2 and EastSt3) have system names that start with IF$dsm.  DCC signal masts always require a layout connection, which can be simulated.

The aspect id for LeftDist Off is -1.  That might be OK 

**EasSt2 Issue**

The destination mast is East32.  It is displaying **Clear Expect Stop**.  

There are two issues.

The aspects.xml file has entries that use **Diverging** as a speed.  There is no such speed in the signalSpeeds.xml file.  I use **Medium** for diverging routes.

The second problem is that AspectMapping entry for **Clear Expect Stop** lists **Stop** and then **Diverging**.  Removing the stop fixes the issue.

When I create a custom signal mast system, I list all of the aspects as possible **advancedAspects**.  Then I assign the **ourAspect** to be used.  If a particular pair is forbidden, then it will only have the Stop.  These are quite rare.

danielb987, 2021-12-10:
@SeaU2 
Welcome to the JMRI project!

PR  #10450 with your xml files and icons has now been merged and will be included in the next JMRI version 4.26.

SeaU2, 2021-12-11:
First of all, me: I am a retired software engineer, mostly "C". Started on
my layout a year ago. Hoping to get full signalling and full automatic
running of at least 6 trains on 3 tracks on a small layout, 2.5mX4m. For
the signals I have an arduino with a sketch to use 17 outputs for various
masts. Studied the JMRI signalling and found the aspect signalling system
to be the most futuristic.

1) System names with IF$dsm .... I have only created this fila as a simulor
test, and not cared about a real layout. Why there is IF$... instead of
LF$... I don't know. And I don't know how to change.

2) LeftDist Off is -1 ...:  hmm ...  wasn't 0 the default off...

3) *Diverging* as a speed ...:  Sorry, my mistake, allthough I thought I
read that Diverging was listed as a speed too someplace.

4) AspectMapping entry for *Clear Expect Stop* lists *Stop* and then
*Diverging:
*Sorry, I really do not know how this works.
   As I could find out, you create an aspectMapping for all
possible advancedAspects, and for each you list all possible aspects of
this mast.
 May be I should think like: What aspects can the advanced mast force our
mast to show. The turnout (or block sensor) will force the *Stop*. Hmm..
no... I really do not understand this. Feel there is missing something in
the documentation.
The Norwegian signal system is quite simple. Actually only the distant
signal is dependant of an advanced signal. There is only 3 signals: Main-3,
Main-2 and distant. A distant signal can be mounted under a Main-3 and a
Main-2. (there is also an avalanche signal)
I would have prefared that "appearance-Main2.xml" and
"appearance-Main3.xml" had none aspectMappings. That would make more sense.
What happens on an end station with one turnout and two end tracks? There
is a Main 3-light on station enter and thats it. For me it seems I have to
make a dummy signal on each end bumper.

But! ... As I see it now ... all <ourAspect>Stop</ourAspect> should be
removed from all <aspectMapping> ? There are none advanced masts that force
stop. Stop is only set by block sensor or turnout (or by controller).  Or?
.... Still think I have missed something.

<specificappearances>
  These I do not quite understand. They are probably 100% wrong.

Kenneth

dsand47, 2021-12-11:
### Aspect IDs

The NMRA standard for DCC aspect ids defines 32 IDs with a value of 0 to 31.  The only specified ID is zero which represents Stop/Danger (UK).  Many of the implementations that I have seen use 31 for Dark.

### Signal mast logic

The JMRI implementation of SML has 2 phases.  

1.  What set of conditions is required so that the signal can display an aspect other than **_danger_**?

When generate/discover is used, the blocks and turnout positions between the current mast and the next mast are used to create the SML.  This is based on the layout topology that is derived from the Layout Editor panel.  The signal masts are placed at block boundaries which is how the signal mast pairs are identified.

Note:  The danger specific appearance from the appearance xml file is used to specify the actual aspect.
 
 
2.  If the mast can be other than **_danger_**, what aspect should be used?

The speed of the SML depends on the speed of the blocks and turnout speeds.  If a SML has **_any_** thrown turnouts, the route will be considered diverging.  If **_all_** of the turnouts, if any, are closed then the route is normal.

The signal mast definition has the signal system and mast type.  This is mapped to an appearance file.  The current aspect of the destination mast is use to find the AspectMapping entry by looking at the advancedAspect names.  This provides a list of ourAspect entries.  These are compared to the current speed and norma/diverging state from the SML to find the best fit from the aspect.xml file.

Once ourAspect has been selected, the appearance icon will be used for the panel and depending on the driver type, the appropriate commands will be sent to the layout.

----

Hopefully this reduces confusion.

SeaU2, 2021-12-13:
>    1. If the mast can be other than *danger*, what aspect should be used?

Stop
What is the "Stop" aspect? Is the program looking for the word "Stop"? Or
perhaps <dccAspect>0</dccAspect>?
I made the mast with 2 stops: "Stop" and "BlockStop", thinking I could use
the same mast as a station exit signal and also as a block signal by using
"disable aspect". They differ by that block-signal having flashing red and
the other having steady red.
I am beginning to think I need to have different appearence-files for
Block- and Main-signals. ???

dsand47, 2021-12-13:
> Stop What is the "Stop" aspect? Is the program looking for the word "Stop"? Or perhaps <dccAspect>0</dccAspect>? I made the mast with 2 stops: "Stop" and "BlockStop", thinking I could use the same [...]

Each appearance xml file has section of specific appearances.
```
  <specificappearances>
    <danger>
      <aspect>Stop</aspect>
    </danger>
    <permissive>
      <aspect>Off</aspect>
    </permissive>
    <held>
      <aspect>Stop</aspect>
    </held>
    <dark>
      <aspect>Off</aspect>
    </dark>
  </specificappearances>
```

Signal mast logic has four keywords for special cases.  This section maps the keywords to the aspect that will be displayed when the keyword is active.  If you disable the Stop aspect, when the "danger" condition exists you will get an error message on the system console.

Is there ever a situation where a specific signal mast has to operate as both a Stop and BlockStop signal mast.  if not, then a BlockStop mast should have its own appearance file.

JMRI does not care about the Aspect IDs.  It just sends what is defined.  The hardware needs to know how to map an ID to the correct LED actions.

SeaU2, 2021-12-13:
> "Stop"? Or perhaps 0? I made the mast with 2 stops: "Stop" and "BlockStop",

Ok, thanks.
Then I have to make some changes and have a Block signal and a Station Exit
signal.
Main2 and Main3 will be station exit, but not quite sure yet if Main2-2 and
Main3-2 can be block-signal and no more needed.
I'll have to ask someone if a block signal exist without a distant signal.
Probably. And not sure if Main3-2 exist without beeing a block signal.
The images are ok.
I might come with new appearance files
Main2, Main3, Main3-2, Main2-2
and
Block2, Block3, Block3-2, Block2-2

Kenneth

SeaU2, 2021-12-13:
Should I change  "Norway2001" to "Norway-2001" inside the xml-files?

dsand47, 2021-12-13:
I would recommend that the names be consistent.

SeaU2, 2021-12-14:
Now the Norwegian signals are working quite ok. (Several block signals
after each other was not working)
Have done some corrections and added Block3 and Block3-2. They are the same
as Main3 and Main3-2 but stop is flashing red. Have removed "BlockStop"
aspect.
Also corrected Aspect/22.gif, 3and2-dark.gif, 2and2-dark.gif.

In my test I have used a Main2 as a dummy on the "End Bumpers". What are
people doing here? We have no special signal on end bumper.

There are probably som error on dark/unlit and specificappearances of which
I don't quite understand yet.

Bug: the flashing icons are not flashing ... yet. May be gif animations
later.

Norway-2001.zip and NorwegianSignals.zip to follow.

regards Kenneth / Sea-U2

danielb987, 2022-01-28:
> [Norway-2001.zip](https://github.com/JMRI/JMRI/files/7710785/Norway-2001.zip)
> [NorwegianSignals.zip](https://github.com/JMRI/JMRI/files/7710787/NorwegianSignals.zip)

@SeaU2 
Have these new files been added to JMRI? We might have missed this.

SeaU2, 2022-01-31:
These were forgotten and I created a request some days ago.  They are in
now.

closed 2022-03-17 by danielb987