JMRI/JMRI #10461: Function panel not working

dougjuk, 2021-12-12:
Issue Type: __Bug__

Loaded the throttle on desktop, I can select a loco and control its speed.  The Function panel is unclickable just does not function.  You cannot even move or select the panel.

I can get focus on the other windows in the throttle but not the function one.

Using windows 11 on a Intel i7 10 gen, 32 meg ram, 1TB samsung ssd, Nvidia 2060 super with up to date drivers.

PanelPro 4.24+Re18b309e8
Java HotSpot(TM) Client VM Oracle Corporation 25.311-b11 mixed mode
Windows 10 10.0 x86

bobjacobsen, 2021-12-12:
This is likely a known problem with that Java version (8u311).  Look in the console (in the JMRI help menu) and if you see an "Exception" message there, that confirms it.

The workarounds include any one of 
 - Install the most recent JMRI test release
 - Wait for JMRI production 4.26 in 10 days or so
 - Install an _earlier_ Java 8
 - Install Java 11

dougjuk, 2021-12-13:
Thanks Bob, Just after sending this in I found if I turned off the option to link with the roster it worked fine again.  Will also look at your other suggestions..

Lionel642, 2021-12-22:
btw this issue with function panel not working sometimes when link to roster option is enabled should be fixed in JMRI latest version. (I think fix was in pull #10004 )

closed 2022-02-20 by stale[bot]