JMRI/JMRI #10487: JMRI Decoder Pro download on Windows 11

Tomslayout, 2021-12-17:
Is there special download I can use to download JMRI on a Windows 11 platform? I tried using the windows 7-10 procedure, but it did not work. Any help you can provide would be helpful.

Thomas Costello
dumpa_hd@yahoo.com,

danielb987, 2021-12-17:
@Tomslayout 
Welcome to the JMRI project!

What error do you get? Any error message?

Tomslayout, 2021-12-17:
I don’t get any error messages. I selected Widows 7-10 to download onto a new computer that has windows 11 I purchased. I have Decoder Pro on older computer which has widows 8 but this computer runs very slow. The Widows 11 I have is 11S (safety). I’m trying to turn of the S but have not figured out how to do that. I’m not sure if shutting off the Saftey program will allow me to down load Decoder Pro. If you have any ideas to get the decoder program onto Windows 11 please advise me. Thanks Tom Costello

danielb987, 2021-12-17:
You need to get out of Windows 11 S mode to be able to install JMRI. Look at this page on how to do it:
https://support.microsoft.com/en-us/windows/switching-out-of-s-mode-in-windows-4f56d9be-99ec-6983-119f-031bfb28a307

closed 2022-02-16 by stale[bot]