JMRI/JMRI #10488: Windows Scaling Factor Not Working

Mark-W-L, 2021-12-17:
Issue Type: __Bug__

I have a custom scale factor set in Windows Display: Change the size of text, apps and other items to 200%. JMRI windows are not applying this in 4.26; it worked in 4.24.

PanelPro 4.26+R381c8dfc32
OpenJDK 64-Bit Server VM Eclipse Adoptium 11.0.13+8 mixed mode
Windows 10 10.0 amd64 [not sure why this says amd, it is an Intel(R) Core(TM) i7-3770K. See Computerinfo.txt]

[systemInfo.txt](https://github.com/JMRI/JMRI/files/7737652/systemInfo.txt)
[session.log](https://github.com/JMRI/JMRI/files/7737653/session.log)
[Computerinfo.txt](https://github.com/JMRI/JMRI/files/7737665/Computerinfo.txt)

danielb987, 2021-12-17:
There is a new setting that might apply to your problem. The [4.26](https://www.jmri.org/releasenotes/jmri4.26.shtml) release note tells:
`Users that have HighDPI scaling set to something different than 100% can now select whenether or not to force Java to use 100% scaling. To change it, go to Edit -> Preferences -> Display. The checkbox has the text Force 100% scaling when the OS uses higher scaling.`

Mark-W-L, 2021-12-18:
If the checkbox 'Force 100% scaling when the OS uses higher scaling' is unchecked and the OS  custom scaling is set to 125% then all linear dimensions are changed by that factor (I note that Windows says this is 'not recommended'). In 4.24 it just made the fonts larger. The font size can be increased in the Display preferences but this does not affect the font sizes in the 'Set Active Profile' start up window for Panel Pro. However, it DOES affect the font sizes in the 'Set Active Profile' start up window for  Decoder Pro! Once JMRI is started the increased font size is applied in all Panel Pro and Decoder Pro windows.

When the status of the checkbox 'Force 100% scaling when the OS uses higher scaling' is changed and Save clicked,  Decoder Pro displays a restart window. Panel Pro does not and when quit is selected it displays the Unsaved Changes dialogue.

danielb987, 2021-12-18:
Which Java version did you use with 4.24 and which Java version did you use with 4.26? There has been a change in Java itself between Java 8 and Java 11.

Mark-W-L, 2021-12-18:
Java 8 with 4.24. I followed the recommendation for 4.26 and downloaded Java 11.

closed 2022-02-16 by stale[bot]