JMRI/JMRI #10533: Webserver view not displaying Slip/Crossover Panel icons

nigelcliffe, 2021-12-28:
As per subject - the Webserver-Browser view of a Panel does not display Slip or Crossover icons.    Other elements work correctly. 

Can be shown with a Panel containing only one icon from the Slip/Crossover/Threeway group.    The summary of the panel on the webserver will show the icon, but actually opening it in a browser and the panel is missing the icon. 

Occurs in JMRI 4.26, and some late 4.25.x test versions, have not checked back to earlier releases.  Observed on two independent Windows machines and one Raspberry PI, covering both Java8 and Java11.

mstevetodd, 2021-12-28:
Thanks for the report, @nigelcliffe 

Please attach an example of a simple panel showing the issue.

nigelcliffe, 2021-12-28:
File as requested below.  It will display on the Panel, but slip not visible on Webserver/Browser view. 

[SlipNotRenderingOnWebServer.zip](https://github.com/JMRI/JMRI/files/7786302/SlipNotRenderingOnWebServer.zip)

silverailscolo, 2022-02-13:
@nigelcliffe can you check the latest build and confirm your slips appear in the web panel?

nigelcliffe, 2022-02-13:
The Slip is behaving properly, but there is an bug in the 3-Way.  

If the 3-way is touched in the Webserver UI, it displays correctly.   
BUT, if it is controlled from (say) the turnout table, it goes through the states of "lower, middle, unknown", rather than "lower, middle, upper".     The Panel view operates correctly.   This means Logix (etc.) cannot control the icon.   

- Nigel

silverailscolo, 2022-02-14:
Thanks Nigel. This tells me the formula I used to determine the icon state from 2 turnout states has a flaw. Will leave this issue open and put this on my (longer term) list, combined with clean up of ``SlipTurnoutIcon.Java``.

silverailscolo, 2022-03-05:
@nigelcliffe Is this now working as expected? If so, please close issue

nigelcliffe, 2022-03-06:
I've downloaded a recent build from Jenkins, and yes, seems to be fixed for 3-way and Double-slip icons.   
Thank you for the work in fixing the problem. 

Issue marked as closed as requested.

closed 2022-03-06 by nigelcliffe