JMRI/JMRI #10553: Incorrect product ID in decoderIndex.xml

chrisorah, 2022-01-08:
**Issue**: Line 21540 has productID="MC1H10421P">
**Correction**: Line 21540 should be productID="MC1H104P21">

The file has this decoder as an OEM Bachman E0Z Command Decoder product number 36-557 as shown [HERE](https://www.themodelcentre.com/36-557-bachmann-e-z-command-1-amp-4-func-21-pin-dcc-decoder-dc-compatible)

This decoder is made by soundtraxx under the product number MC1H104P21 as shown [HERE](https://www.ebay.com/itm/254581718758)

Product ID just has a typo. On a side note, it would be nice to have this added under SoundTraxx > MC Decoder Folder as they are one in the same.

bobjacobsen, 2022-01-08:
That file is made automatically from the contents of the individual decoder definition files.  Can you please figure out which file has the original error?

chrisorah, 2022-01-09:
Hello, 

Please see this file: https://github.com/JMRI/JMRI/blob/master/xml/decoders/SoundTraxx_OEM_36-557_MC1.xml

On Line 20 - <model model="Bachmann (soundTraxx MC1) 36-557" numOuts="4" numFns="14" connector="other" productID="MC1H10421P">

_Link to decoder model for reference: https://soundtraxx.com/products/dcc-mobile-decoders/mc1h104p21/_

closed 2022-01-11 by danielb987