JMRI/JMRI #10555: Common method to store and load script engine to XML

danielb987, 2022-01-09:
@bobjacobsen 
Now when we give the user the ability to select which script engine to use for scripts, we will now get several places where we need to store and load these settings. Logix, LogixNG, and others.

If I have understood it correctly, Nashorn has been removed in Java 15. This means that users will need to migrate their Nashhorn settings to Graal.

I suggest that we create two common store and load methods that stores and loads this setting. The load method can then check if Graal is available and Nashorn is not, and then return Graal instead of Nashorn even if the user has selected Nashorn.

danielb987, 2022-05-13:
@bobjacobsen 
How is the status of Graal and a possible Nashorn replacement? Have you heard any news of either the Nashorn replacement project or about the problems we had with Graal?

bobjacobsen, 2022-05-13:
We're a long way from being able to use Graal Python as a option for JMRI's scripting.  Javascript is _much_ closer to working; with only minor changes, we could use that as a Nashorn replacement now.  (Of course, getting people to switch to a Graal VM, etc, would be a burden)

I reopened this Issue because having a centralized way of opening scripts is a good idea on its own, and is a necessary step toward better (eventual) scripting.

danielb987, 2022-09-29:
It would be good to keep it open for now.

closed 2023-02-28 by stale[bot]

bobjacobsen, 2023-02-28:
I think it's OK to close it.  It'll be a while before JMRI moves forward to a Java version where it'll really be relevant, and I'm sure it'll come up again then.